RMerl/asuswrt-merlin.ng #34: build failure at protect_srv

azdps, 2017-10-12:
I attempted to build RT-AC3100 from your source code today Oct 11th but it results in the following build failure:

`rm -rf kernel_header
obj-prelibs:
busybox
shared
nvram_arm_7114
libbcm
libupnp_arm_7114
libbcmcrypto
make parallel=true  busybox shared nvram_arm_7114 libbcm libupnp_arm_7114 libbcmcrypto
make[4]: Entering directory '/root/asuswrt-merlin.382/release/src/router'
"CLM In /root/asuswrt-merlin.382/release/src-rt-7.14.114.x/src/../4365/src/wl/clm/src/ NOT CREATING new clm_blob ..."
make -C protect_srv/lib
make[5]: Entering directory '/root/asuswrt-merlin.382/release/src/router/protect_srv/lib'
cp -f prebuild_arm/libptcsrv.so libptcsrv.so
cp: cannot stat 'prebuild_arm/libptcsrv.so': No such file or directory
Makefile:28: recipe for target 'all' failed
make[5]: [all] Error 1 (ignored)
make[5]: Leaving directory '/root/asuswrt-merlin.382/release/src/router/protect_srv/lib'
install -D protect_srv/lib/libptcsrv.so /root/asuswrt-merlin.382/release/src-rt-7.14.114.x/src/router/arm-uclibc/stage/usr/lib/libptcsrv.so
install: cannot stat 'protect_srv/lib/libptcsrv.so': No such file or directory
Makefile:3764: recipe for target 'protect_srv-stage' failed
make[4]: *** [protect_srv-stage] Error 1
make[4]: Leaving directory '/root/asuswrt-merlin.382/release/src/router'
Makefile:1420: recipe for target 'obj_prelibs' failed
make[3]: *** [obj_prelibs] Error 2
make[3]: Leaving directory '/root/asuswrt-merlin.382/release/src/router'
Makefile:392: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/root/asuswrt-merlin.382/release/src-rt-7.14.114.x/src'
Makefile:3756: recipe for target 'bin' failed
make[1]: *** [bin] Error 2
make[1]: Leaving directory '/root/asuswrt-merlin.382/release/src-rt-7.14.114.x/src'
Makefile:3898: recipe for target 'RT-AC3100' failed
make: *** [RT-AC3100] Error 2
'
It appears a libptcsrv.so is not present in directory prebuild_arm. Looks like you have the Hnd one though.

RMerl, 2017-10-12:
I have it locally, I just didn't properly push it to Github  (got filtered by gitignore) - will do so in a moment.

However this won't help you, as we're missing the dhd24.o wifi driver for the RT-AC3100.  You could always try to experiment by copying the RT-AC88U version in the wl_arm_7114 RT-AC3100 folder, no idea if that will work however.

The first 382 GPL release for SDK7114 has been somewhat of a mess so far.

closed 2017-10-12 by RMerl

azdps, 2017-10-12:
The latest 382 source code released by ASUS for the RT-AC3100 contains the folder wl_arm_7114 and includes both dhd.o and dhd24.o files. I'm somewhat confused. Aren't these files the ones that are needed instead of coping the ones released with the RT-AC88U source code?

RMerl, 2017-10-12:
Mine doesn't have the dhd24.o file.  I believe Asus did reupload the GPL3100 archive at some point, my copy was probably the original uploads.  I'll redownload it and take a look.

commit f65be725 (2017-10-12): protect_srv: push ARM prebuilt files (got filtered by gitignore) (close #34)