RMerl/asuswrt-merlin.ng #36: SDK7114: build error at u2ec

azdps, 2017-10-12:
Build failure at u2ec. The repository is also missing prebuilt arm u2ec. The route/u2ec folder only contains the HND one. Here's the error:

```
-e    u2ec   
make[4]: Entering directory '/root/asuswrt-merlin.382/release/src/router/u2ec'
make[4]: *** No rule to make target 'usbsock.o', needed by 'u2ec'.  Stop.
make[4]: Leaving directory '/root/asuswrt-merlin.382/release/src/router/u2ec'
Makefile:7833: recipe for target 'u2ec' failed
make[3]: *** [u2ec] Error 2
make[3]: Leaving directory '/root/asuswrt-merlin.382/release/src/router'
Makefile:392: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/root/asuswrt-merlin.382/release/src-rt-7.14.114.x/src'
Makefile:3756: recipe for target 'bin' failed
make[1]: *** [bin] Error 2
make[1]: Leaving directory '/root/asuswrt-merlin.382/release/src-rt-7.14.114.x/src'
Makefile:3898: recipe for target 'RT-AC3100' failed
make: *** [RT-AC3100] Error 2
```

azdps, 2017-10-12:
Replacing the makefile and adding prebuild/u2ec from the RT-AC3100 382 source code solved this problem. The current makefile you have will obviously needed to adjusted to include the regular arm u2ec binary.

RMerl, 2017-10-15:
You should close issues rather than erasing/renaming them.

closed 2017-10-15 by RMerl