RMerl/asuswrt-merlin.ng #37: SDK7114: missing several arm specific close source binary files

azdps, 2017-10-12:
Missing the following:

- asuswebstorage/prebuild_arm/asuswebstorage
- inotify/prebuild_arm/inotify
- webdav_client/prebuild_arm/webdav_client
- dropbox_client/prebuild_arm/dropbox_client

Could be more I'm still looking.

```
make[4]: Entering directory '/root/asuswrt-merlin.382/release/src/router/asuswebstorage'
cp -f /root/asuswrt-merlin.382/release/src-rt-7.14.114.x/src/router/asuswebstorage/prebuild_arm/asuswebstorage asuswebstorage
cp: cannot stat '/root/asuswrt-merlin.382/release/src-rt-7.14.114.x/src/router/asuswebstorage/prebuild_arm/asuswebstorage': No such file or directory
Makefile:31: recipe for target 'all' failed
make[4]: [all] Error 1 (ignored)
make[4]: Leaving directory '/root/asuswrt-merlin.382/release/src/router/asuswebstorage'
make[4]: Entering directory '/root/asuswrt-merlin.382/release/src/router/inotify'
cp prebuild_arm/inotify inotify
cp: cannot stat 'prebuild_arm/inotify': No such file or directory
Makefile:8: recipe for target 'all' failed
make[4]: [all] Error 1 (ignored)
make[4]: Leaving directory '/root/asuswrt-merlin.382/release/src/router/inotify'
touch webdav_client/stamp-h1
make[4]: Entering directory '/root/asuswrt-merlin.382/release/src/router/webdav_client'
cp -f /root/asuswrt-merlin.382/release/src-rt-7.14.114.x/src/router/webdav_client/prebuild_arm/webdav_client webdav_client
cp: cannot stat '/root/asuswrt-merlin.382/release/src-rt-7.14.114.x/src/router/webdav_client/prebuild_arm/webdav_client': No such file or directory
Makefile:28: recipe for target 'all' failed
make[4]: [all] Error 1 (ignored)
make[4]: Leaving directory '/root/asuswrt-merlin.382/release/src/router/webdav_client'
touch dropbox_client/stamp-h1
make[4]: Entering directory '/root/asuswrt-merlin.382/release/src/router/dropbox_client'
cp: cannot stat 'prebuild_arm/dropbox_client': No such file or directory
Makefile:2: recipe for target 'all' failed
make[4]: *** [all] Error 1
make[4]: Leaving directory '/root/asuswrt-merlin.382/release/src/router/dropbox_client'
Makefile:5487: recipe for target 'dropbox_client' failed
make[3]: *** [dropbox_client] Error 2
make[3]: Leaving directory '/root/asuswrt-merlin.382/release/src/router'
Makefile:392: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/root/asuswrt-merlin.382/release/src-rt-7.14.114.x/src'
Makefile:3756: recipe for target 'bin' failed
make[1]: *** [bin] Error 2
make[1]: Leaving directory '/root/asuswrt-merlin.382/release/src-rt-7.14.114.x/src'
Makefile:3898: recipe for target 'RT-AC3100' failed
make: *** [RT-AC3100] Error 2
```

azdps, 2017-10-12:
Try building the RT-AC3100. There's more problems.

RMerl, 2017-10-12:
I'm not at the stage of implementing or testing that model yet.

azdps, 2017-10-12:
Okay, stop spamming problems until you get to this stage.

closed 2017-10-15 by RMerl