RMerl/asuswrt-merlin.ng #38: rc/wan.c add_multi_routes define problem

azdps, 2017-10-13:
add_multi_routes is defined in rc/rc.h and also rc/wan.c. I noticed the issue when setting DUALWAN=n in target.mak. You will see that in wan.c at line 161 add_multi_routes is defined again. Since it's defined in rc.h and rc.h is included in wan.c there's no need to define it again.

RMerl, 2017-10-15:
Same in Asus's GPL drop.

RMerl, 2017-10-15:
Actually, it's only declared in rc.h, and it gets defined either in wan.c (single WAN) or private.o (dual WAN).  I don't see anything wrong with that.

azdps, 2017-10-16:
This is the error I get with DUALWAN=n in target.mak:

```
 [rc] CC natnl_api.o
 [rc] CC erp_monitor.o
 [rc] CC rc
prebuild/private.o: In function `add_multi_routes':
(.text.add_multi_routes+0x0): multiple definition of `add_multi_routes'
wan.o:wan.c:(.text.add_multi_routes+0x0): first defined here
collect2: ld returned 1 exit status
Makefile:570: recipe for target 'rc' failed
make[4]: *** [rc] Error 1
make[4]: Leaving directory '/asuswrt/release/src/router/rc'
Makefile:7479: recipe for target 'rc' failed
make[3]: *** [rc] Error 2
make[3]: Leaving directory '/asuswrt/release/src/router'
Makefile:382: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/asuswrt/release/src-rt-7.14.114.x/src'
Makefile:3686: recipe for target 'bin' failed
```

RMerl, 2017-10-17:
It's a conflict between private.o (closed source) and wan.c (open source).  To compile without this error you would need the source code to private.o, so you can recompile it with DUALWAN=n to remove its conflicting function.  Therefore nothing that can be done about it.

closed 2017-10-17 by RMerl