RMerl/asuswrt-merlin.ng #40: missing closed source /usr/sbin/arpstorm & wl_server_socket

azdps, 2017-10-16:
I noticed both /usr/sbin/arpstorm and /usr/sbin/wl_server_socket are missing. It appears arpstorm was previously available with ASUS 380 GPL. arpstorm appears to be part of networkmap but as a separate compiles binary file.

wl_server_socket appears to have never been released in any GPL. I could be wrong. Not a clue on what it exactly does.

Both of these files can be found in the released 382 firmware for the RT-AC3100 under /usr/sbin folder. Although can not be found in the 382 source code.

azdps, 2017-10-16:
I manually added it to the RT-AC3100 382 source code under networkmap. Looks like arpstorm is going to be closed source. Not sure of the value of arpstorm anyways whether it's actually being used. By the way II have a fully functional working firmware now using the ASUS's 382 source code for my RT-AC3100.

RMerl, 2017-10-16:
I know arpstorm is used by ATE (which is mostly used for manufacturing/debugging), don't remember if it's used by the userspace code however.

First time I hear abotu wl_server_socket, so no idea what it's for, and whether it's used or not.

azdps, 2017-10-17:
Interesting to see wl_server_socket in /user/sbin in the officially released ASUS 382 firmware for the RT-AC3100. The only reason I noticed it was because I was  checking the comparing all the binary files in the officially released firmware verses what was produced by manually building the firmware from scratch.

There's several files in the RT-AC86U source code with sw_server_socket. This this also true for the 380 source code.

[https://goo.gl/L3Dgrz](url)

RMerl, 2017-10-17:
wl_server_socket might get compiled in only when compiling from proprietary sources.  It's part of Broadcom's SDK, however Asuswrt doesn't seem to use it, so it's not necessary (and has never been part of the GPL drop either).  Asuswrt relies on the wl userspace tool when it needs to access the wireless driver.

Broadcom's SDK does include various components that aren't used by Asuswrt.  This is especially true with HND, which includes all sort of unused components, including debugging tools and SysV Init scripts that don't get called anywhere by Asuswrt.

arpstorm might be needed to ensure that networkmap gets quickly populated at boot time.  I'll look into recovering it from both SDK7114 and HND firmware releases.

RMerl, 2017-10-17:
Asus are deliberately NOT including arpstorm if using prebuilt binaries rather than compiling from sources.  Not sure why, my only guess might be for legal reasons (as technically, arpstorm might be considered a malicious executable?).

```
ifneq ($(wildcard $(SRCBASE)/router/networkmap/*.c),)
        install -D arpstorm $(INSTALLDIR)/usr/sbin/arpstorm
        $(STRIP) $(INSTALLDIR)/usr/sbin/arpstorm
endif
```

closed 2017-10-17 by RMerl

azdps, 2017-10-17:
Yes I noticed that in the makefile. I changed mine to include it anyways. Looking at the 380 source code, arpstorm.c does show the fuctions etc. I did see some debug in it but there might be some useful information as to what it actually does.
[]https://github.com/RMerl/asuswrt-merlin/blob/master/release/src/router/networkmap/arpstorm.c(url)