RMerl/asuswrt-merlin.ng #41: [Q] About KRACK

S0und, 2017-10-16:
https://www.krackattacks.com/

At this point do we have to wait for ASUS to release a fix, or can we expect Merlin to fix the vulnerability?

RMerl, 2017-10-16:
From what little info I have so far (I haven't had time to study the disclosure yet), this only affects STA mode, not AP mode.  Therefore in theory, the only updates needed for the router would be for Repeater/Bridge modes.

There's a chance this might require a newer nas exe from Broadcom, in which case I can't do anything until Asus issues an update.

No idea yet if the firmware-included wpa_supplicant is used in Repeater mode.

Ryujinra, 2017-10-18:
@RMerl Evidently, there is an AP-side patch to hostapd that can address unpatched clients; see here: https://github.com/RMerl/asuswrt-merlin/issues/1511#issuecomment-337542831

RMerl, 2017-10-19:
Asuswrt does not use hostpad.

Ryujinra, 2017-10-19:
Would it still be possible to disable the retransmission of EAPOL-Key frames in wpa_send_eapol() in wpa_auth.c?

themiron, 2017-10-19:
asuswrt don’t use wpa_supplicant for anything wireless on bcm models. it’s for wired 802.1x there

TecumsehSamp, 2017-10-23:
Do you know if wpa_supplicant is used in AP mode for wireless clients?

RMerl, 2017-11-02:
HND platform fixed with commit 71f5b52aadfd51ac93cace52c3f4d3937640e7be.

Still no updated SDK available for SDK7114.

RMerl, 2017-12-04:
SDK7114 fixes merged with fd20998, which fixes the last remaining models on the 382 branch.

closed 2017-12-04 by RMerl