RMerl/asuswrt-merlin.ng #49: clean code and wrong nvram value

odkrys, 2017-11-12:
Advanced_OpenVPNClient_Content.asp
```
			document.getElementById('edit_vpn_crt_client_crl').value = vpn_crt_client4_crl[0].replace(/&#10/g, "\n").replace(/&#13/g, "\r");
			document.getElementById('edit_vpn_crt_client_extra').value = vpn_crt_client3_extra[0].replace(/&#10/g, "\n").replace(/&#13/g, "\r");  <----- ?
			document.getElementById('edit_vpn_crt_client_extra').value = vpn_crt_client4_extra[0].replace(/&#10/g, "\n").replace(/&#13/g, "\r");
			break;
```

defaults.c
This value seems related with strange nat rule.
```
{ "atcover_sip_ip", "192.168.1.1
```
```    0     0 DNAT       tcp  --  any    any     anywhere             192.168.1.1          tcp dpt:www to:192.168.50.1:18017
```

RMerl, 2017-11-22:
18017 redirection is part of the WEBREDIRECT feature, it`s unrelated to this nvram value you quote.

closed 2017-11-22 by RMerl

odkrys, 2017-11-22:
192.168.1.1 destination value is right value on rt-ac86u?
default lan address is 192.168.50.1.

commit 5802d1ed (2017-11-22): webui: remove extra line leftover from copy/paste (close #49)