RMerl/asuswrt-merlin.ng #50: [Tor] Evaluate migrating to 0.3.x.x

RMerl, 2017-11-16:
Consider migrating to Tor 0.3.x.x as it should be fairly mature by now.

@blackfuel - any gotchas you've found since moving to 0.3.x.x, config file changes required, etc...?

blackfuel, 2017-12-10:
There are no gotchas AFAIK.  I'm using the latest stable Tor without any changes and it's running fine.

Tor 0.3.x.x now has optional support for zstd and lzma compression libraries.  If you've got these libraries installed, Tor will now say that you're using these libraries, in torlog.  Example:
```
... running on Linux with Libevent 2.0.21-stable, OpenSSL 1.0.2n, Zlib 1.2.11, Liblzma 5.2.3, and Libzstd 1.3.2.
```

*LZMA package*
https://tukaani.org/xz/xz-5.2.3.tar.gz

*ZSTD package*
https://github.com/facebook/zstd/archive/v1.3.2.tar.gz

I've also got the Tor geoip and geoip6 databases installed in the router firmware.  This could make a difference for when Tor selects "the best" guard node.  I think Tor will automatically choose a guard that is outside your jurisdiction (i.e. different country).

Sorry I didn't see this message sooner.  Not receiving the Github messages.

RMerl, 2017-12-10:
Thanks.  For 382.2 I will stick with the updated 2.8 version, and will consider a move to 0.3 with 382.3.

RMerl, 2019-02-26:
Done with 42b75587b640f520800b22cdf0ea40edeb926885 .

closed 2019-02-26 by RMerl