RMerl/asuswrt-merlin.ng #52: xt_geoip: the bundled version can't handle 64bit kernel with 32bit userspace

noblehng, 2017-11-20:
Because the [struct xt_geoip_match_info](https://github.com/RMerl/asuswrt-merlin.382/blob/master/release/src-rt-5.02hnd/kernel/linux-4.1/include/linux/netfilter/xt_geoip.h#L36) has different size under 64bit and 32bit, one will get an error saying "x_tables: ip_tables: geoip.0 match: invalid size 160 (kernel) != (user) 96" in dmesg when trying to use the geoip xtables addon on RT-AC86U, which has a 64bit kernel and 32bit userspace.

Upstream has been properly handling this since [2008](https://sourceforge.net/p/xtables-addons/xtables-addons/ci/7aae90da5a891936a768c2d011bf0c297df9221d/). Could it be updated to a more recent version? Or one should build the kernel module and userspace part themself then use bind mount?

RMerl, 2017-11-22:
I don't maintain the geoip module since it's not used by Asuswrt. Asus added the module a few years ago but it never got used, probably was part of plans that were dropped later.

bigcat133, 2017-11-30:
i have same problem， when use iptables get that：
`iptables: Invalid argument. Run `dmesg' for more information.

x_tables: ip_tables: geoip.0 match: invalid size 160 (kernel) != (user) 96`

noblehng, 2017-12-03:
@RMerl So this could be closed as asus bug, I guess?

@bigcat133 Because CONFIG_MODVERSIONS is not set in the kernel config, and xtables_addons doesn't have other dependencies, so it can be compiled directly against the kernel source without Module.symvers.

Get toolchains [here](https://github.com/RMerl/am-toolchains/tree/master/brcm-arm-hnd), set up `PATH` and `LD_LIBRARY_PATH`, then get the kernel source for RT-AC86U in this repo, finally build the modules in xtables_addons source dir with:
```
XA_ABSTOPSRCDIR=$PWD ARCH=arm64 CROSS_COMPILE=aarch64-buildroot-linux-gnu- make -C $KDIR M=$PWD/extensions
```
where `$KDIR` is your kernel source dir.

RMerl, 2018-01-30:
Closing since this is related to a component that isn`t used by the firmware (I suspect Asus initially planned to do something with GeoIP and ultimately dropped those plans, without removing the code from the repo).

closed 2018-01-30 by RMerl

SylphiaWindy, 2018-08-22:
@noblehng I tried to build xtables_addons v2.14 and uploaded xt_geoip.ko to /tmp,
loaded with insmod, but iptables says "No chain/target/match by that name"
lsmod shows xt_geoip was loaded. Any thing I'm missing?