RMerl/asuswrt-merlin.ng #53: Client name/icon user modification not allowed after 30ish entries

mattheweli, 2017-11-27:
After approx 30 clients have been personalized, by changing their names/icons in either DHCP server client list or Wireless MAC Address filter or Wirelss LAN Guest MAC Address list pages, then this function stops working and the field "Name" would remain "New Device" and the stadard icon is mantained.

The icon/name of these clients anyhow get modified after the devices connect to the router for the first time and the name take a sort of the identification of the device (in my case a Huawei P10 took "HUAWEI_P10" as its Name); also the icon is personalized after the device connect for the first time

RMerl, 2017-11-27:
Nothing I can do about it - Broadcom enforces size limitations on nvram settings with the new SDK, and the code validating/enforcing it is closed source.  The DHCP list is now limited to 1000 characters max.

closed 2017-11-27 by RMerl

mattheweli, 2017-11-27:
Oh ok; sorry to hear that. I don't know if this could be helpful, but I dug in this problem and it seems the clientList structure gets inconsisten after 31 entries. I enclosed a picture here to be more clear.

[https://i.imgur.com/NG8bbwy.png](url)

I can say, the first 31 entries are named correctly; after that, the 32nd and more, are not referenced correctly in the structure. Is this related to the limit you mentioned?