RMerl/asuswrt-merlin.ng #54: Manually assigned DHCP addresses do not honor blank domain name

kwloafman, 2017-12-02:
I like to keep the host addresses on my LAN static by using the router's MAC assignment rather than sprinkling /etc/hosts files all over the place. This router is adding ".local" to the hostname, making the address something like "ceres.local". Is there something I need to set to get it back to clean behavior? This extra typing and misnaming is annoying. It should be just "ceres"!

RMerl, 2017-12-28:
Can't reproduce here.  When I set a static lease and I don't provide a domain, doing an nslookup properly returns just the hostname without any domain.

How are you observing this behaviour?  Note that .local is used by Avahi/mDNS, so if your resolver also queries mDNS, then it might be appending the .local TLD to it, which is intended behavior.

kwloafman, 2017-12-29:
It's probably due to the VPN.  When I set Accept DNS Configuration to
exclusive, I think it does force the use of mDNS.  I have not had time to
check further.

closed 2017-12-29 by RMerl