RMerl/asuswrt-merlin.ng #58: Linker error when BWDPI=n

blackfuel, 2017-12-10:
When compiling 382.x with `BWDPI=n` for RT-AC68U, there is a reference in the closed source Trend Micro piece `httpd/web_hook.o` to `libbwdpi.so`.  It fails to link properly with `BWDPI=n`.  
```
make -C httpd
make[2]: Entering directory '/home/blackfuel/asuswrt-merlin.382/release/src/router/httpd'
 [httpd] CC httpd.o
 [httpd] CC httpd
prebuild/web_hook.o: In function `DPIisSupport':
(.text+0x224c): undefined reference to `dump_dpi_support'
collect2: ld returned 1 exit status
Makefile:264: recipe for target 'httpd' failed
make[2]: *** [httpd] Error 1
make[2]: Leaving directory 
```

AND, if you try to create a stub function for `dump_dpi_support()`, it gets invoked repeatedly and negatively impacts performance.

RMerl, 2017-12-10:
Nothing I can do about that, since web_hook is closed source.  You will have similar issues with a lot of settings that are tied to code located within the closed source objects.  Same reason why I can't enable IPSEC support on the RT-AC68U either, as the Tuxera kernel modules will fail to load due to the kernel symbol changes involved.

RMerl, 2018-01-30:
Closing since this is not a bug, but a limitation tied to the closed source components.

closed 2018-01-30 by RMerl