RMerl/asuswrt-merlin.ng #59: Question About DFS

Stellarspace, 2017-12-18:
How come on the Verizon Fios G1100 router, DFS channels are enabled by default? The router can select it automatically if it thinks it would be the best channel, but users can't select the DFS channels. Users can't enable or disable DFS, but users can select channel #165. On Fios Network Extenders, users can choose to enable or disable DFS, but you can't select any of the DFS channels once again and users can't select channel #165. But on an Asus RT-AC66U B1 router, you can't select channel #165, the router can't use any DFS channels, not even automatically. You can't enable or disable DFS either. I have multiple access points on channel #136 because the AP set it automatically to that channel. Why can't ASUS support enabling and disabling DFS? You wouldn't be able to select it, but it would be able to set it to a DFS channel if it is the best channel. How come ASUS clients don't work with DFS either? I have a PCE-AC68 and it doesn't work with channel #136. I know about the FCC regulation, but how does it apply to this case?

RMerl, 2017-12-19:
The issue tracker is only for development purposes.  If you need technical support or have a question, please use the support forums at https://www.snbforums.com/forums/asuswrt-official.51/ instead.

closed 2017-12-19 by RMerl

Stellarspace, 2017-12-19:
@RMerl Why can't you add the ability to select channel #165 and the ability to enable or disable DFS and the router would automatically select the channel if it is the best channel?

RMerl, 2017-12-19:
Wifi code is closed source and cannot be modified.

Stellarspace, 2017-12-19:
@RMerl Are the wireless drivers that ASUS provides also closed source?

RMerl, 2017-12-19:
Yes, both the wireless driver and all the code that configures/manages wifi.