RMerl/asuswrt-merlin.ng #65: ipv6: routing and firewall issues

noblehng, 2018-01-22:
When setting up ipv6, the address is added to the wan interface with prefix using the ip command, and this will cause the kernel to automatically create a route of that prefix on the wan interface, by default and by standard. For example: [tunnel](https://github.com/RMerl/asuswrt-merlin.ng/blob/master/release/src/router/rc/services.c#L1887), [static](https://github.com/RMerl/asuswrt-merlin.ng/blob/master/release/src/router/rc/wan.c#L2268) and [dynamic](https://github.com/RMerl/asuswrt-merlin.ng/blob/master/release/src/router/rc/udhcpc.c#L1259). This route will conflict with a similar route on the lan br0 interface, which is the correct one because the subnet should route through our lan not the wan. As a result, ipv6 of the clients using the router become very flaky.

For newer kernel like the `HND_ROUTER` ones, the `noprefixroute` option can be used. For older ones, a workaround would be setting the prefix of the wan interface to 128. The workaround can be apply on the webui for static ipv6 address now, but not for dynamic ones. [Here](https://github.com/systemd/systemd/issues/1520#issuecomment-147358931) has a good explanation.

Besides routing, there may also be a issue about firewall rule of ipv6. For security reason, inbound connection to ipv6 clients of the router should not be forwarded by default. But a default `ACCEPT` rule is added [here](https://github.com/RMerl/asuswrt-merlin.ng/blob/master/release/src/router/rc/firewall.c#L3959) and [here](https://github.com/RMerl/asuswrt-merlin.ng/blob/master/release/src/router/rc/firewall.c#L5101). The `filter_wl_default_x` doesn't seem to be used these days, so a explict accept rule is added before drop target of the forward chain. Note that the important icmpv6 already has its own rules set for forwarding.

themiron, 2018-01-23:
you need different/more specific subnet for lan, either static or via dhcpv6 pd.
if you have no such, you can use 128 as wan prefix length.
if you have no such with dhcpv6/slasc, use passthrough ipv6 type.

noblehng, 2018-01-23:
That is a workaround for now, I personally am using the 128 prefix trick for my tunnel. But it did take some time to find the cause and workaround it when I decided to look into the problem last weekend. I think it would be better to handle this out of box by the firmware codes.

And I thought using dhcpv6 relay/passthrogh is harder for firewalling, less manageable and secure. This could also be worked around for now, by using custom script to delete the route, or replace the `/tmp/dhcp6c` symbol link of `rc` to our own for handling odhcp6c state change though.

themiron, 2018-01-23:
what kind of tunnel do you use? if tunnelbrocker.net - it allows to get additional routed /48 for any purpose, for example /64 from it you can use for lan subnet.
firewall is working for ipv6 passthrough in the same way, as for different ipv6 wan types.

I see no any need to change something, use scripts, replace links, it just works, not?
if somehting doesn't work for you, please specify your exact environment and setup.

noblehng, 2018-01-23:
It works for me now, but it needs some tinkering, and it might not work for other user that has no clue of these things. So I think it can be improved by dong what networkd of systemd does, that is just use 128 prefix for older kernel and use noprefixroute option  for newer ones when configuring ipv6 address for wan interface in the `rc` code.

Use script and replace links are for someone geting dhcpv6 from upstream but don't want to use passthrough for downstream, so that they can explict manage their clients inside the router.

themiron, 2018-01-23:
please provide usual real life case which can't be handled with current logic and settings, but will be handled with your suggested changes.

noblehng, 2018-01-23:
Simply using /64 on both wan and lan will not work without knowing the /128 trick, with the change it can handle it transparently.

Using passthrough, you can not easily assign static ipv6 addresses to your devices without using another device.

themiron, 2018-01-23:
simply using same /64 on both wan and lan is configuration error if not in passthrough mode. nothing more, same as using same ipv4 network on both lan and wan. it's up to user.
using passthrough router does much more to handle same subnet on lan & wan correctly.

trae32566-w, 2018-01-23:
Just for future reference, typically the way you would configure IPv6 for home use, you would be given a /128 (the "client IP" for Hurricane Electric), and then have a /64 (or other prefix) **routed to** that address. You should not be using the same IP on WAN and LAN, and as @themiron noted, that is misconfiguration. 
As well, this goes for DHCPv6, where you would also be handed a /128 via IA_NA, and then a /64 (or other prefix) via DHCPv6-PD.

noblehng, 2018-01-23:
Sorry, my bad. I' m splitting a /64 myself using 6in4, it was a hack. So forgot about the routing issue. dhcpv6 without dhcpv6-pd is also possible, but that will also be a hack.

themiron, 2018-01-23:
sharing dhcpv6 subnet without pd is also supported, with passthrough mode. no configuration is required at all.
the only left is firewall issue

samip5, 2018-04-02:
Could we get an iptables firewall GUI on the Web UI of the router, or some sort of IPv6 firewall that can be accessed from the Web UI as well?

closed 2020-09-21 by RMerl