RMerl/asuswrt-merlin.ng #70: ctf: broke iptables POSTROUTING MASQUERADE

V-E-O, 2018-01-26:
I have one AC56U with latest beta3 firmware, also tested with stable 380.* firmware that when NAT acceleration enabled, port forwarding iptables POSTROUTING MASQUERADE rule seems broken.

Here is iptables rules that I used.
`iptables -t nat -A PREROUTING -p tcp --dport 161 -j DNAT --to 192.168.1.4:161`
`iptables -t nat -A POSTROUTING -p tcp -d 192.168.1.4 --dport 161 -j MASQUERADE`

I cannot setup port forwarding directly in Merlin, as Merlin router is not on my default route table on server 192.168.1.4, when the server receives the incoming TCP packet, src-ip is Internet IP, another route is later choosen but route from Merlin.

So I used second rule in my scenario to alter incoming packets src-ip from Internet IP to router IP(192.168.1.1).

Howerver when NAT CTF acceleration enabled, this rule is broken.

Here is Wireshark TCP session screenshot on server 192.168.1.4.
[image]

You can see src-ip was altered to 192.168.1.1 in SYN, RST packets, but abnormal when data transmitting (PSH).

Here is normal TCP session after with NAT CTF acceleration disabled.

[image]

BTW, some ports like 21 function well in both config :(
Thanks!

RMerl, 2018-01-30:
CTF bypasses parts of iptables, which is why manual port fowards won't work.  This is a technical limitation of CTF.

For port forwards to work, you need to mark them so they will bypass CTF.

closed 2018-01-30 by RMerl

V-E-O, 2018-01-31:
@RMerl Is it good idea to add a configurable option to mark which ports can bypass CTF? Currently I cannot find a way, like setting Port Forwarding in WebUI and then apply absent second rule: iptables -t nat -A POSTROUTING -p tcp -d 192.168.1.4 --dport 161 -j MASQUERADE

V-E-O, 2018-01-31:
This rule can make packets forwarding to 192.168.1.* to bypass CTF.
Then POSTROUTING works normal.

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination         
MARK       all  --  192.168.1.0/24       192.168.1.0/24       MARK xset 0x1/0x7

However, some other configs but port forwarding set this mark, I need to look up.

RMerl, 2018-01-31:
mark bit 1 is used for CTF bypass.  Other bits are used for other things, such as QoS, NAT loopback, etc....

V-E-O, 2018-01-31:
@RMerl Yeah, thanks. Just can not figure out which config automatically enables this rule (maybe VPN ipsec related). Nevermind, I manually enabled it in my own script.