RMerl/asuswrt-merlin.ng #78: Stale IP address in VPN/VPN Status

kwloafman, 2018-02-05:
The IP address in the status display is not changing as the VPN address changes.  It is showing the same address it started with, yet the address has changed a few times since the start as shown by ipecho and others.  And no, a Refresh has no effect.

RMerl, 2018-02-08:
The IP only gets looked up on first access, to avoid hammering Google's STUN servers.  Figuring out when to refresh it might be tricky if I still want to avoid initiating too many access attempts to Google's servers.

kwloafman, 2018-02-09:
No need to refresh on a schedule, just when the Refresh button is hit would
be sufficient.

RMerl, 2018-02-09:
That would also imply refreshing it every time that page gets loaded (Refresh merely reloads that page).  Since VPNStatus is the first page of that section, it means every time you accessed the VPN section, an STUN lookup would be done.  I'd rather avoid that if possible.

kwloafman, 2018-02-09:
Maybe a button next to the IP address that only refreshes the IP?

awdsmirk, 2018-03-10:
As far as hammering STUN servers are concerned, how about just creating a pool or increasing the size of the current pool (if any) of STUN servers to use for the VPNStatus? There's quite a few lists of public STUN servers available with a quick Google search. This one looks reputable: https://www.voip-info.org/wiki/view/STUN
I guess what I'm saying is to spread the load around so as to not hammer any one company's/organization's STUN servers. 
Plus isn't STUN supposed to be unreliable since it uses UDP? So hammering shouldn't be an issue since there's no guarantee of reliability, right?

RMerl, 2018-03-11:
Some users will object to having their router connect to a bunch of random IPs (and those using OpenVPN clients tend to be the most paranoids).  I stuck to these stun servers because Asus were already using them in their getrealip.sh script.