RMerl/asuswrt-merlin.ng #79: httpd starts on port 80 although not configured in settings

erikdewildt, 2018-02-10:
Hi,

I have explicitly configured my RT-AC88U router to only use HTTPS as i'm running an NGINX reverse proxy on port 80. Since i've upgraded to 384.3_beta2 the httpd daemon starts at boot and occupies port 80.

> admin@RT-AC88U-0888:/tmp/home/root# netstat -lnp
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
... ... ...
tcp        0      0 127.0.0.1:80            0.0.0.0:*               LISTEN      397/httpd

> admin@RT-AC88U-0888:/tmp/home/root# ps axf | grep http
 2648 pts/1    S+     0:00          \_ grep http
  396 ?        Ss     0:54 httpds -s -i br0 -p 8443
  397 ?        Ss     0:00 httpd -i br0

[image]

Thanks!

RMerl, 2018-02-11:
Asus does it on purpose for models with AiHome support, probably because the service relies on it.  As a workaround, move httpd to a different port.

```
nvram set http_lanport=81
nvram commit
service restart_httpd
```

themiron, 2018-02-12:
@erikdewildt fyi, due closed source, asusnatnl, bwdpi blocking and aicloud redirect may not work properly after port change. this needs more updates from asus

RMerl, 2018-02-12:
As long https is enabled tho he should be fine, those services should normally use https if http is listed as disabled.

themiron, 2018-02-12:
right with doubts about asusnatnl, otherwise why httpd was kept running then?

RMerl, 2018-02-12:
I don't know.  But since virtually nobody uses asusnatnl (it's for their AiHome solution), I doubt it would be a problem for @erikdewildt .

erikdewildt, 2018-02-17:
@RMerl Indeed, that is no problem fo me. Your workaround works just fine. Thanks for the support!

closed 2018-02-17 by erikdewildt

dreamwraith, 2018-06-01:
Would it be possible to add a configurable option to change this setting in the menu?