RMerl/asuswrt-merlin.ng #83: R7000: LED to display status only

blshkv, 2018-02-15:
Thank you very much for a wonderful work!

I miss one feature from the original firmware where LED would not blink all the time and reflect the connection status only.

The closed feature I found is the "LED stealth mode" but it switches off lights completely.

Would you consider implementing such feature please?

RMerl, 2018-02-15:
Running Asuswrt-Merlin on a non-Asus router is illegal, due to proprietary licenses used by firmware components.

closed 2018-02-15 by RMerl

blshkv, 2018-02-15:
Oops, I did not realise it and will be erasing it. FYI, the firmware is publicly available without any warnings at http://xvtx.ru/xwrt and it even links to this repository.

RMerl, 2018-02-16:
I know.  Xvortex even stole the content of my previous website for his own, and has rejected all my previous requests for a GPL release, and for the removal of Asus's proprietary components.