RMerl/asuswrt-merlin.ng #84: 384.3-Beta3 cannot change anything in system settings after clean install or nvram reset

patcfly, 2018-02-15:
Clean install or reset nvram
log into admin \ settings
attempt to change any setting
expected result: settings change
actual result: shown error message "Specified time interval can not overlap."

Additional info: drop down menus for daylight savings time are not shown.

patcfly, 2018-02-15:
Update: i downgraded to latest stable, configured time options, changed admin pw, enabled ssh and saved. Then upgraded using UI to 384.3-Beta3 and now the daylight savings dropdown menus are available/work and I am able to make and save changes.

closed 2018-02-22 by RMerl

bradanlane, 2018-03-15:
This problem is present in 384.3 "release".

Changing the Time Zone to "GMT" disables the fields and validation for "DST time zone changes starts" and "DST time zone changes ends".

Any Time Zone with DST support renders the additional settings fields with empty inaccessible drop downs.