RMerl/asuswrt-merlin.ng #88: Make SMB 2.0 default

Josverl, 2018-02-21:
Please change the default to SMBv2=Enabled , and SMBv1=Disabled ,
and clarify the description for the setting to show better what the setting is.
Current:
The current default enables SMBv1, and disables SMBv2. 

Proposed Minimal:
Enable SMB2 protocol 
default: Yes  | Options Yes - No

Recommended:
SMB2 protocol version 
default: V2   | Option : V1 - V2 

Rationale:
I see the defaults introduce a risk for attacks such as WannaCry that make use of vulnerabilities in the SAMBA and Windows implementation of SMBv1, and use a downgrade approach. 

My (foolish) assumption was that the default SMBV2=Off meant to read that the system was running SMBv3, only later I found out that this was SMBV1.

I do not think that SMBv1 is a safe default anymore. While it might allow simpler connection by older clients, the risk that SMBV1 introduces does not outweigh a secure default. 
Users that need SMBv1, can still enable it,

RMerl, 2018-02-22:
The reason for not enabling SMBv2 by default is the 10-12% performance hit it brings on lower-end models.  The RT-AC86U was an exception because of its beefier hardware, so it could be set as enabled by default.

With Windows gradually phasing out SMBv1 I am willing to reconsider it for older models as well, however I must take into account the amount of user complains I will get for the performance drop.

Making the labeling for this option isn't a problem, and would make sense.

RMerl, 2018-02-22:
An additional factor that`s making me pause: if I set SMB2 as the default protocol, it means that anyone upgrading from the stock firmware to my firmware will be unable to access any router share until they reboot their computers (or restart the Workstation service) - Windows cannot renegotiate established SMB protocols.  That opens the door to a lot of potential support requests.

Microsoft realized this themselves, which is why Windows 10 will not remove SMBv1 support on an upgrade - it will only uninstall it if it`s unused for a certain period of time, or if you are doing a fresh install rather than an upgrade.

RMerl, 2018-02-22:
Reworked the current settings implementation, with commit beb8eedb01a9d43850ef9b3b8c92ba358d831013

1) Replaced setting with a dropdown
2) Available options are SMBv1, SMBv2, and SMBv1 + SMBv2
3) Default is now SMBV1 + SMBv2 (which is the same as the original "Enabled" setting)

If Asus ever migrates to Samba 3.6, then I will switch the default value to SMBv2.

closed 2018-02-22 by RMerl

unsynaps, 2018-02-22:
Now if only Asus will update to Samba 4 so we can get SMBv3 and encryption.

RMerl, 2018-02-23:
I've managed through a lot of hacks and patches to get Samba 4 to cross-compile.  The resulting build is a complete clusterfuck of files.  I doubt we'll see Samba 4 on Asuswrt anytime soon.