RMerl/asuswrt-merlin.ng #90: Remote Syslog Forwarding

Xboarder56, 2018-02-23:
Perviously ASUSWRT-Merlin supported remote syslog destinations through the web interface. I can enable this through the CLI interface but this will not be maintained through a reboot. Will this feature be coming back or is it discontinued ?

RMerl, 2018-02-23:
The setting is now on the System Log page itself.

Xboarder56, 2018-02-23:
Ohh my bad sorry about that didn't notice it got switched around. When an IP is enter for the remote log server where does this go?

RMerl, 2018-02-23:
It will be sent to whichever IP you enter on that page.

Xboarder56, 2018-02-23:
Sorry I mean to say the config file that will be updated with the IP.

RMerl, 2018-02-23:
There's no config file.  Asuswrt uses Busybox's syslogd, which can only be configured through command line switches - there's no config file.

Xboarder56, 2018-02-23:
Ohh okay thanks for the help.

closed 2018-02-23 by Xboarder56