RMerl/asuswrt-merlin.ng #92: SNMP on HND generates flood of "kernel: ****** Error: MDIO Access Returned FAIL"

RMerl, 2018-02-27:
net-snmp is currently having issues on the HND platform, flooding the log with these errors:

```
kernel: ****** Error: MDIO Access Returned FAIL! CL22 Access: Op: Read; PortAddr: 0x00; DevAddr: 0x07; Reg_in: 0x08070000; Reg_out: 0x1807ffff
```

Attempted update to 5.7.3 (the latest version) without any change.  I suspect one of the modules might be incompatible with the HND platform.

Temporary solution is to disable SNMP support for the RT-AC86U.

ruddj, 2018-05-09:
Will SNMP support for RT-AC86U be returning in a future build? 
I use it with Cacti to monitor traffic, CPU and memory usage on router.

RMerl, 2018-05-10:
I don't know.  Depends on if Broadcom makes any future change to the network driver that would make it compatible.  Personally, I doubt it.

closed 2020-09-21 by RMerl