RMerl/asuswrt-merlin.ng #94: Time Remaining on guest wifi network does not work

timkgh, 2018-03-01:
v384.3 on AC68U running in AP mode
If I set Time Remaining on Guest wifi network to 1 hr, it doesn't tick down, it's stuck at 59 min. and the network is not shut down after 1 hr, so not just a GUI issue.
This used to work in previous versions, it looks like a regression in the new code.

timkgh, 2018-06-10:
@RMerl any progress on this issue?

crkinard, 2018-06-10:
Is it still an issue on 384.5?

timkgh, 2018-06-10:
Can someone who runs it in router mode confirm the issue is there too? I have mine in AP mode, not sure if related to the mode or not.

CHRONOMASTER, 2018-07-14:
I have this issue too in **Wireless router mode** on my RT-AC68U.

I noted this issue in 384.5 and 380.69_2 ( there was other earlier versions but I didn't write them down )

Daopp, 2018-12-22:
Same issue here in 384.6 on ASUS RT-3200, Wireless router mode. Setting a time out of 3 hrs will only cause it to get stuck on 2 hrs 59 mins indefinitely.

AppliedReality, 2019-01-04:
This looks to me to be an Asus issue.  Asus 3.0.0.4.384_45149 on AC68U behaves in a similar way.  The countdown stops at 1 minute.  It should be noted that the guest SSID goes off the air.  The device is functioning as expected but the status report is wrong.

closed 2020-09-21 by RMerl