RMerl/asuswrt-merlin.ng #95: JFFS partition unmounted.

crkinard, 2018-03-01:
Came across an issue on my new RT-AC86U running 384.3.

If you tell it to format the JFFS partition on next boot it is errors out somewhere (can not find anything in the logs) causing the partition to not mount. Only fix is to reflash the router then it mounts no problem... till you tell it to format it again.

Psudo order of things I tried:
Unboxed and powered on.
Flashed to current ASUS firmware and rebooted.
Flashed 384.3. Rebooted. Factory defaulted. Rebooted.
Administration -> System -> "Format JFFS partition at next boot" & "Enable JFFS custom scripts and configs". Both to yes and rebooted.
Setup SSH and all. Connected with WinSCP.
Tried to create folder in /jffs said it was read only. Checked "df -h" and Tools tab.unmounted in both.
Administration -> System -> "Format JFFS partition at next boot" to yes. Rebooted.
Still unmounted.
Reflashed 384.3. /jffs mounted and a bunch of folders (i assume the system uses and all that).
Administration -> System -> "Format JFFS partition at next boot" to yes. Rebooted.
/jffs unmounted

RMerl, 2018-03-08:
Can't reproduce on my own RT-AC86U.  I set it to format at next boot, rebooted, and after the reboot the jffs partition was correctly mounted, with the previous content gone from it, and am also able to write to it.

In the past some users reported that they sometime needed to reboot a second time after formatting, for the partition to get remounted correctly.  That's probably why the reflash fixed it for you - it generated a second reboot.  There's nothing in the flash process that should affect the jjffs partition.

crkinard, 2018-03-10:
Quite... annoying but the double restart seems to work.

closed 2018-03-10 by crkinard