RMerl/asuswrt-merlin.ng #116: Weird QoS behavior. Upload/download reversed.

iMocW, 2018-03-27:
When traffic go through OpenVPN client, LAN side download traffic will show as upload traffic &vice versa. This is really frustrating, I have to disable QoS. Because I have 200/20 D/U WAN link, this "bug" will result in 20Mbps(D&U) max on OpenVPN tunnel.
Can you try run QoS on WAN interface exclusively even though it will recognize OpenVPN traffic as General traffic? Or the best solution is reverse the now situation so it can recognize all traffic correctly.(Yes, now it can recognize what traffic is running through the OpenVPN, just reversed)

modemboy, 2018-03-28:
This is an Asus bug, Merlin can't fix.
https://www.snbforums.com/threads/vpn-traffic-shows-as-uploads-in-qos-asus-bug.30030/

RMerl, 2018-05-07:
Should be resolved with 1c86fbc5c2398895a7b086a78738f7373cd611d9.

closed 2018-05-07 by RMerl

iMocW, 2018-05-14:
No, this fix is not for this issue.