RMerl/asuswrt-merlin.ng #117: Formatting JFFS erases DHCP manual assignement configuration

Diaoul, 2018-03-28:
And potentially other settings too as I noticed that my SSH authorization key was gone too...

I previously had a RT-AC66U and now a RT-AC86U (so ng firmware). I don't recall having this behavior back then, I seem to recall that jffs was disabled by default and only used for a few things.

Anyways, if this is the expected behavior, first, sorry for the noise, then please put a warning somewhere or a tooltip. Anything that makes it more obvious that (some?) settings are going to be erased.

Thanks for the awesome features you bring to the firmware, keep up the good work!

crkinard, 2018-04-01:
On the 86 some of this stuff is saved in /jffs/nvram now.

RMerl, 2018-04-05:
This is indeed intended, as Asus recently moved some settings to the JFFS partition to resolve issues specific to that Broadcom platform.

I added a warning that will be shown whenever someone sets the format option to "Yes" (commit 7e207b6bc0c5d5ea8b765983c17d0176484912d5).

closed 2018-04-05 by RMerl