RMerl/asuswrt-merlin.ng #129: 384.4_2 -- error on changing image for dhcp client

thezoggy, 2018-04-11:
LAN > DHCP Server
clicking on one of the icons in the the list once you have something assigned to a static ip, I noticed at times it wouldn't let me change the image. Looking at the console I see the javascript error:
`SyntaxError: missing ) after argument list[Learn More] Advanced_DHCP_Content.asp:1:44`

An example device where this is failing is one of my computers which has a MSI motherboard, the device looks to contain values that I would suspect is not quoted/escaped properly:

Device:  `MICRO-STAR INT'L CO.,LTD`

Darkrulerz, 2018-04-17:
Hi @thezoggy, I'm a programmer/developer who has recently installed asuswrt-merlin on my devices and now I'm trying to help out.

Perhaps you can give me some more info so I might be able to create a pull-request later on.

Do you see the error in the console right after opening the page or does it only occur after pressing apply on the popup that shows up where you can edit the image?

thezoggy, 2018-04-17:
only after I click on to assign an image for the device.

thezoggy, 2018-05-17:
@RMerl any chance you've taken a look at this ?

RMerl, 2018-05-17:
No, didn't have time to yet.

thezoggy, 2018-11-05:
just an update, this issue is also in latest: `384.7_2`

thezoggy, 2018-11-20:
I see you put in a fix for this issue,
https://github.com/RMerl/asuswrt-merlin.ng/commit/8f48643026a5e406b5d8a15218742d5357dfbd54

download 384.8b1 to try out

thezoggy, 2018-11-20:
I can confirm it works as expected now, thanks!

closed 2018-11-20 by thezoggy