RMerl/asuswrt-merlin.ng #135: AC87u - Most JS files don't load in chrome on mobile unless "desktop mode" is used

Darkrulerz, 2018-04-17:
Router AC87u - 384.4_2

On my phone, using Chrome (v65) after logging in most JS files error out because their content is set to
```
<HTML><HEAD><script>top.location.href='/Main_Login.asp';</script>
</HEAD></HTML>
```
It seems a JS-file is requested but HTML is returned with a status 200 containing a (JS) redirect-attempt to the main-page.

A quick scan of the code led me to believe that https://github.com/RMerl/asuswrt-merlin.ng/blame/091a90f3480e9f406f3b88c40db85c9e4398df74/release/src/router/httpd/httpd.c#L443 is the culprit.

I'm not sure why a request for a JS file would end up there...

More stuff I've tried:
- Worked fine in 380.xx
- Enabling "Desktop mode" on mobile allows the page to load normally.
- Switching to mobile-emulation in desktop Chrome works fine.
- Also tried with my girlfriends phone, same result...

Let me know if I can do anything to help debug/reproduce this issue

RMerl, 2018-05-18:
That code section is only called when trying to connect two AiMesh nodes together, it's not called for regular end-user webui access.

Darkrulerz, 2018-11-28:
Got fixed somewhere along the way. Works fine now. Thanks

closed 2018-11-28 by Darkrulerz