RMerl/asuswrt-merlin.ng #137: Tinc VPN

Diaoul, 2018-04-22:
*I've opened this issue on the legacy tracker but as it has been shut down I cannot reopen there.*

Tinc is a meshed VPN service.
I'd like to be able to setup a Tinc VPN between multiple AsusWRT routers. It's rather easy to setup and I think it would be a great addition to the firmware.
There's already plenty VPN supported, tinc would be great for completeness and a good fit for a router.
Although I agree this is and edge case, it seems to me that TOR is quite the Edge case too and has been added to the firmware (don't know if this is Asus' doing or Merlin's).

Would you accept a PR for this feature request? If not, could you point me directions on how to cross-compile tinc so I can manually add it myself? I already have cross-compilation knowledge.

Thanks and keep up the good work!

themiron, 2018-04-22:
does installing tinc from entware suit your needs?

Diaoul, 2018-04-22:
I assume the binary would work. I'd like to avoid having to install entware
and would prefer reuse the built-in libs for compilation of I go down this
road.

A GUI integration would be much better though, I'd like to avoid
compilation or entware.

themiron, 2018-04-22:
things I've mentioned are not advanced stuff, it's just tinc basics, it's ,uch simplier and very different from openvpn.
manual ceriticate adding actually is more complex than join, where is all the info aout upstream node. limiting to 6 is redundant, tinc handles node certs in runtime indirectly.
also, need some way to change/select the (multiple) nodes to connect to.

personally I've been using tinc for ages, sure, would be great to see it integrated into asuswrt-merlin. sufficienly integrated, I mean. @RMerl opinion?

RMerl, 2018-04-22:
I don't want to implement an additional VPN technology.  Proper support would also require implementing the webui, implementing policy-based routing, and after that maintaining it with security updates.  The additional workload cannot be justified, therefore I have no intention of adding any new VPN technology to the router.  OpenVPN and IPSEC covers pretty much everyone's need.

closed 2018-05-03 by RMerl