RMerl/asuswrt-merlin.ng #138: Enhanced user script support

fantom-x, 2018-04-22:
For a long time now I was missing the support for user scripts via the use of folders like **firewall-start.d**, etc. This way all those additional features that require the use of the user script functionality could just drop their scripts into an appropriate folder without a risk of overwriting the existing files. 
I am offering to implement and test this if it has a chance of being useful and be merged.

RMerl, 2018-04-22:
I don't want to make things more complicated for end-user.  Changing the user script behaviour at this point by going for an init.d-similar setup would make things very confusing for new users, who would be facing both the previously implemented scripts in addition to a folder-based system.

At this point in this firmware's age, I believe it's too late for such a radical design decision change.

fantom-x, 2018-04-22:
Good point, but I was not suggesting a replacement. It would be backwards compatible and not mutually exclusive. The handler would generate a new script under **/tmp/** and put the original one on the first line followed by alfabetically ordered **ng** scripts.

samip5, 2018-05-29:
Idea is so if that kind of setup exists, use it and if not failback to the one in use now?

RMerl, 2018-09-10:
Don't have any plans to change the way custom scripts are currently implemented.

closed 2018-09-10 by RMerl