RMerl/asuswrt-merlin.ng #140: Time machine backup size limited to 900GB

mieszkomularczyk, 2018-04-25:
I have RT68U with 384.4_2 (stable) installed. If I go to : USB Applications --> Time Machine --> Time Machine Volume Size, I can set it to max 900GB even though my drive is 4T and empty. Please allow changing that size above 0.9 TB.

RMerl, 2018-05-03:
There is no hardcoded max size, the limit is calculated based on the volume size.

```
if(!validator.rangeAllowZero(document.form.tm_vol_size, 0, parseInt(availSpace), document.form.tm_vol_size.value))
                return false;
        else
                document.form.tm_vol_size.value = document.form.tm_vol_size.value*1024;
```
Do you have multiple partitions on your disk or multiple plugged disks by any chance?

mieszkomularczyk, 2018-05-03:
Hi @RMerl ,
In my case:
I have single 4TB drive (no other drives) connected to RT68U. Single ext4 partition.  And now I see the problem. The algorithm right now is not correct - it works nice when you setup TM for the first time - then of course GUI should allow up to free capacity. In my case mac backups already exists and they use 800GB and I have still 900GB free on that drive. Thus I should be allowed to set TM size to 1.5GB, even though remaining capacity is 900GB.
The algorithm should be: maxTM size = size of currect TM backups + free capacity. Probably first component is hard to calculate, thus I would allow entering TM size up to drive size with a warning msg only if it exceeds current free capacity.

mieszkomularczyk, 2018-05-14:
btw, @RMerl : meanwhile I hacked it by logging to ssh and setting tm_vol_size manually to a higher value in NVRAM :)

RMerl, 2018-05-18:
Someone else would have to tackle this one as I have no way of testing it.

kolobock, 2018-05-18:
Confirming the same on RT-AC56U:
Changing the TM volume size allows setting a free disk space as a maximum value even it is a higher already.

closed 2020-09-21 by RMerl