RMerl/asuswrt-merlin.ng #144: release/src-rtk-819x missing

royalmustard, 2018-05-03:
Maybe there is a good reason for it, but I am really desperate because my RP-AC55 is not working.

TheUntouchable, 2018-05-03:
Perhaps your device is not working because its not supported?

Asuswrt-Merlin officially supports the following router models:

    RT-N66U
    RT-AC66U
    RT-AC66U_B1 (same firmware as the RT-AC68U)
    RT-AC56U
    RT-AC68U  (including revisions C1 and E1)
    RT-AC68P (same firmware as RT-AC68U)
    RT-AC68UF (same firmware as RT-AC68U)
    RT-AC87
    RT-AC3200
    RT-AC88U
    RT-AC3100
    RT-AC5300
    RT-AC1900 (same firmware as RT-AC68U)
    RT-AC1900P (same firmware as RT-AC68U)
    RT-AC86U (starting with version 382.1)

(Note: the U, R and W variants are all supported, as they are the exact same hardware and firmware, only different marketing SKUs or different case color)

 

The following models are no longer officially supported:

    RT-N16

RMerl, 2018-05-03:
It's not missing, it's simply not supported by this project.

closed 2018-05-03 by RMerl