RMerl/asuswrt-merlin.ng #146: Support GCM ciphers in OpenVPN UI.

CODYQX4, 2018-05-11:
There are issues I have with some VPN providers that require disabling cipher negotiation, and I want to use AES-256-GCM with them. With negotiation on, they set AES-128-GCM. With it off, I can only choose CBC, even though the router supports GCM if negotiated.

RMerl, 2018-05-11:
If you change the negotiation cipher settings on the router to only contain AES-256-GCM (it default to multiple ciphers), what happens then?

If it still falls back to AES-128-GCM then something might be misconfigured on their end.  I can try to see if providing a GCM cipher is allowed by OpenVPN in the old legacy setting.

CODYQX4, 2018-05-11:
If I force AES-256-GCM with no fallback, I can't connect and I get these errors:
```
WARNING: 'link-mtu' is used inconsistently, local='link-mtu 1586', remote='link-mtu 1602'
WARNING: 'cipher' is used inconsistently, local='cipher BF-CBC', remote='cipher AES-256-CBC'
WARNING: 'keysize' is used inconsistently, local='keysize 128', remote='keysize 256'
PUSH: Received control message: 'PUSH_REPLY,redirect-gateway def1,dhcp-option DNS 10.9.0.1,dhcp-option DNS 10.8.0.1,route 10.31.0.1,topology net30,ping 5,ping-restart 30,ifconfig 10.31.0.10 10.31.0.9,peer-id 1,cipher AES-128-GCM'
Error: pushed cipher not allowed - AES-128-GCM not in BF-CBC or AES-256-GCM
```

If I force AES-256-GCM with a fallback of AES-256-GCM, I can't connect and I get these errors:
```
PUSH: Received control message: 'PUSH_REPLY,redirect-gateway def1,dhcp-option DNS 10.9.0.1,dhcp-option DNS 10.8.0.1,route 10.31.0.1,topology net30,ping 5,ping-restart 30,ifconfig 10.31.0.10 10.31.0.9,peer-id 1,cipher AES-128-GCM'
Error: pushed cipher not allowed - AES-128-GCM not in AES-256-CBC or AES-256-GCM
OPTIONS ERROR: failed to import crypto options
ERROR: Failed to apply push options
Failed to open tun/tap interface
```

Their provided config includes these relevant lines. These disable cipher negotiation and push the GCM cipher in a legacy way. If I use this config via an OpenVPN client, everything is fine. If I try to force that cipher line into custom configuration, I get errors in the log about the auth digest being inconsistent.
```
cipher AES-256-GCM
ncp-disable
```

I can connect with Viscosity. I get warnings, but the connection does work, though I'm not sure if the security has dropped or if those warnings are just a glitch. If this is a broken state, then they are definitely doing something wrong on their end. Despite the warnings, the log indicates that it is using SHA-512 and AES-256-GCM.
```
Outgoing Control Channel Authentication: Using 512 bit message hash 'SHA512' for HMAC authentication
Incoming Control Channel Authentication: Using 512 bit message hash 'SHA512' for HMAC authentication
WARNING: 'link-mtu' is used inconsistently, local='link-mtu 1550', remote='link-mtu 1602'
WARNING: 'cipher' is used inconsistently, local='cipher AES-256-GCM', remote='cipher AES-256-CBC'
WARNING: 'auth' is used inconsistently, local='auth [null-digest]', remote='auth SHA512'
Control Channel: TLSv1.2, cipher TLSv1/SSLv3 DHE-RSA-AES256-GCM-SHA384, 2048 bit RSA
Outgoing Data Channel: Cipher 'AES-256-GCM' initialized with 256 bit key
Incoming Data Channel: Cipher 'AES-256-GCM' initialized with 256 bit key
```

RMerl, 2018-07-06:
Implemented with 929073f9cac4263072bf2c79b150abe3503e0f2d .

closed 2018-07-06 by RMerl