RMerl/asuswrt-merlin.ng #150: 385.0 - RT-AC87U - After upgrading the router still says the version is 384.4_2

oittaa, 2018-05-22:
As the title says, after flashing the newest firmware RT-AC87U_384.5_0.trx (SHA256 9eb8021418881a693a04d4dceaa43ea2167704da5e9583e45a1e59e978700ab5) the web interface still shows 384.4_2 as the version and the "A new version of firmware is now available." symbol keeps flashing.

https://i.imgur.com/ZUytuOJ.png

oittaa, 2018-05-24:
@MacQwerty Thanks! It's not a very elegant workaround, but it worked.

technomooney, 2018-06-06:
be sure you dont have any usb drives plugged in and it should upgrade... i have the same model router

oittaa, 2018-06-06:
I didn't have any USB devices attached. Only couple of Ethernet cables are connected in addition to the power source.

ookonbekend, 2018-06-27:
FYI, this worked for me,
First eject the USB disk and then updated the firmware.

RMerl, 2018-07-25:
The issue tracker is only for development purposes.  If you need technical support or have a question, please use the support forums at https://www.snbforums.com/forums/asuswrt-official.51/ instead.

closed 2018-07-25 by RMerl

oittaa, 2018-07-25:
What do you mean with a comment like that? This is clearly about a bug in your software.

kevinmarchant, 2019-02-04:
> be sure you dont have any usb drives plugged in and it should upgrade... i have the same model router

Thanks technomooney - this caught me out on the 384.7_2 to 384.9 upgrade.  Ejecting the USB stick and upgrade went without hitch.  Useful to know !

isa220, 2019-02-13:
I do factory reset with JFFS and config backup, then I need flash 384.9 twice to complete firmware upgrade (after first flashing version still 384.5). it's no USB stick. same for 384.5 - I need to flash it twice to apply.