RMerl/asuswrt-merlin.ng #151: Default DDNS updater for namecheap (and possibly others) doesn't use https

kaysond, 2018-05-24:
If you select the namecheap ddns updater, you'll see something like this in the logs:
```
May 23 11:48:10 start_ddns: update WWW.NAMECHEAP.COM namecheap, wan_unit 0
May 23 11:48:10 ddns_update: ez-ipupdate: starting...
May 23 11:48:10 ddns_update: connected to dynamicdns.park-your-domain.com (104.219.249.157) on port 80.
May 23 11:48:11 ddns_update: request successful
May 23 11:48:11 ddns_update: asusddns_update: 0
May 23 11:48:11 ddns: ddns update ok
```
It's connecting to the namecheap update server via http, despite https being supported (see https://dynamicdns.park-your-domain.com/getip). Sending the api key in plain text could lead to an attacker redirecting your domain to any server. Presumably it was set up this way because ez-ipupdate doesn't support SSL. In fact it seems that ez-ipupdate is not really maintained anymore.

An easy workaround is to run the custom script found here, but requires jffs to be enabled and so wouldn't work by default: https://github.com/RMerl/asuswrt-merlin/wiki/DDNS-Sample-Scripts

I think for security it would be better not to have namecheap (or any others, really) available as options if they have to use ez-ipupdate. Perhaps an alternate solution could be developed using curl and/or wget since it seems they're both available.

RMerl, 2018-07-06:
Re-implementing the entire DDNS system is beyond the scope of this project.  You should ask Asus to update their implementation to use a more modern client that supports SSL.

closed 2018-07-06 by RMerl

kaysond, 2018-07-06:
I'm happy to do that. Who's the best person/dept to talk to...?

RMerl, 2018-07-06:
networking_support (at) asus (dot) com.