RMerl/asuswrt-merlin.ng #154: 384.5 - WiFi devices aren't communicating between each other

guidomedina, 2018-05-27:
After I upgraded from `384.4_2` to `384.5` on my `RT-AC3200` the WiFi devices aren't communicating between each other.

I have a couple of Linux machines which I SSH between one another and I can't anymore, I then connected to the router directly and SSH into each and it works fine.

the0ne, 2018-06-07:
I experience a similar issue on my RT-AC88U since a while / several firmware versions.
I tried to flash the original ASUS firmware but the problem occurs there as well.

Very annoying as I use a HTTP-Server on an ESP32 (WiFi) to control my shutters using mobile phones (WiFi) but the mobile phones cannot "see" the ESP32 whereas from my PC it works at the same time.

Most interesting: if I disconnect the mobile phone from the WiFi and connect to the router via VPN then the connection works again...

sebastianhaba, 2018-06-11:
I have similar problem. I have yamaha amplituner connected by wifi to my network. I can visit website hosted by this device directly from ac88u but not from other devices in my network. Even from devices connected by cable, but I can connect to my network by openvpn and magically I can access device without problems.

the0ne, 2018-07-04:
Just noticed that by setting
"Set AP Isolated" to **"Yes"**
under http://router.asus.com/Advanced_WAdvanced_Content.asp

-> which obviously is **not** what I want (but just for **testing**)
-> then access from devices connected via cable (e.g. PC) works **more reliable** than with having this set to "No"

Asus seriously needs to fix this - does anybody have some idea on how to let some technicans of Asus know **directly** ?
Probably some first or 2nd level support will not understand the impact of this...

Segfaul198, 2018-07-12:
I've noticed this as well starting a couple of revisions back but it got fairly severe in 384.5 for some reason. During my trouble shooting I've noticed a lot of incomplete arp entries. These also seem to correspond to wireless devices that cannot communicate with others and either just joined the network or where rebooted.

guidomedina, 2018-11-25:
Yeap, but then it could be anything from Asus GPL for specific hardware to RMerlin:
- **RT-AC3200** with 384.5 and 384.6 had the issue.
- **RT-AC2900** with 384.7_2 doesn't have the issue.

ryannathans, 2018-11-25:
How do you know RT-AC2900 with 384.7_2 doesn't have the issue?

RT-AC5300 has had the issue on every version since 384.5 and probably earlier. The problem is it seems to "fix" itself sometimes and then come back - usually when something reboots.

ryannathans, 2018-11-25:
[image]

After manually patching the ARP table - still no success.

Seems as though I can't see any packets at all from the specified device.

guidomedina, 2018-11-25:
Because I bought a RT-AC2900 a week ago?

ryannathans, 2018-11-25:
I can see the (ARP) packets being sent back from 192.168.0.6 but 192.168.0.5 isn't receiving anything back

[image]

guidomedina, 2018-11-25:
I bought the RT-AC2900 a week ago, RT-AC3200 and RT-AC5300 both have TriBand, maybe there is a pattern for which some devices have the issue and others don't.

ryannathans, 2018-11-25:
My point is that the issue is intermittent, so you might think it's fine and it comes back

jcduss59, 2021-06-01:
Hello, I got a RT-AC86U with Merlin 386.2_4
I got the same kind of problem.
I saw that devices on wifi band 2.4Ghz cannot access 5Ghz band.

Smart Connect is disabled. Both 5ghz and 2.4ghz has same name (SSID)
Due to my house configuration I need to force my printer on 2.4ghz band (on 5ghz signal is instable , too far from router, big wall). To do this I do Wireless mac filter on 5ghz band.

My computer work well on 5ghz (best bandwith).

From the router, I can ping both My printer and my computer
From my computer I can ping and ssh my router  etc but not my printer.

if I ping my printer (on 2.4ghz) from computer on (5ghz) what I see from a router tcpdump is that, arp request from computer get no reply. 
I guess arp (and maybe other traffic) is not propacated from wifi bands while it's the same network range.... Not normal.

I cannot also ping anaything (IoT stuff) on 2.4Ghz band from a device on 5ghz band.

There is definitely a bug behind this behaviour.

How could I help you to troubleshoot this ?