RMerl/asuswrt-merlin.ng #155: 384.5 - RT-AC86U - Router crashing intermittently for no apparent reason.

dbmajor, 2018-05-28:
Hi there, 

As described, the router keeps crashing on a regular basis and I can't figure out why.

Log reasons include:
- Unable to handle kernel paging request at virtual address 10101464c81pgd
- Unable to handle kernel paging request at virtual address f1000499d102699fpgd
- dhd_detach(): thread:dhd_watchdog_thread:246 terminated OKFPM Pool 1: invalid token 0x23b10000 freedkthread_should_stop detected in wfddhd_detach():
- dhd_detach(): thread:dhd_watchdog_thread:246 terminated OKFPM Pool 1: invalid token 0x230a0000 freedFPM Pool 1

Router is running:
- RT-AC86U Merlin Firmware 384.5
- RAM at 75%
- Processors fluctuate, but are idle most of the time
- DDNS
- DHCP static assignments
- Traffic Monitoring
- Adaptive QoS
- SAMBA server with 2-3 users
- A 4TB HDD attached to USB 3.0
- A 4GB swap partition on a dedicated USB stick
- PPTP, OpenVPN and IPSec VPN Servers (one of each)
- DLNA server with the database file on the 4TB HDD
- SmartConnect Wifi at full power
- Entware installed via SSH
- 4 ethernet devices attached - 2x Gigabit and 2x 100 Mbps

Thanks for your help and please let me know if you need more information.

Cheers

Syst3mSh0ck, 2018-06-08:
All the 'dhd' calls are for the Dongle Host Driver however the kernel paging issue is a much deeper problem which suggests a memory management issue, possibly indicating a hardware problem. Have you tried removing the USB HDD/Flash? If this doesn't work you could try another flash to 384.5, reboot, factory reset, reboot then configure and reboot once more. Not sure if all the reboots are neccessary but I like to be safe. Failing that you could try downgrading to an earlier version to see if this is a regression.

dbmajor, 2018-07-06:
> The dhd process and everything tied to it is proprietary and outside of my
> control. Nothing I can do about it if there's actually any issue there.

Thanks heaps for that.

I've narrowed it down to keeping the QoS app analysis turned off.

If I turn that on, it crashes frequently.

Thanks for your help.

Cheers,
Josh