RMerl/asuswrt-merlin.ng #156: Allow multiple SSH keys

tuxpeople, 2018-05-28:
Currently, the GUI only allows to save 1 SSH Key. If I enter more than one, none will be saved. I can add more to  /home/root/.ssh/authorized_keys manually, but the GUI still only shows one.

Please allow saving multiple public keys via GUI.

RMerl, 2018-06-06:
You should be able to provide multiple keys.  I know I used to have two keys myself a long time ago (one DSA and one EC), however I haven't retested it lately.

tuxpeople, 2018-06-20:
While I can't be 100% sure not to be the problem, it looks like it's not working. I added multiple keys directly in the file but I can only see one in the WebUI. And (not sure on this) it looks like he's not using all of them. Which is weird. I now added only 1 on the WebUI and that one is working. But someone else should also test this to see if this si a bug or my fault. btw., my keys are of different lenghts.

nullcake, 2018-07-07:
@tuxpeople 

This works as expected.

Whatever you enter into the WebGUI ends up in /tmp/home/root/.ssh/authorized_keys

<pre>
admin@router:/tmp/home/root# cat .ssh/authorized_keys
ssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAABAQCz/CrgeRvQhkq8rSnzzgAMjxxgbe+H1FDa8/aeKfRLwjtJarWdxsXcleNbFmGfebXof4qu1pAQCBEROT7jStIJCU+RMojLT2nKDJltEFqOFXQfCm0ukSwVjDOI3ABbVsa6JOMcxSRMmWh5MXCT4R2eY/U6nKWhn9YValg+LMlu//RrlORBAWrHtCZ9ZDz6+NAkdkE+I2sv7sMkq5KYihkp/eOSlvxHp2j1RKCPk6U3GHY82pA34kPrHyiZFQjZM21dvwwIOPW0ZjydB9H7tlkcWkfecQjQG+B/441Zi8twa7PdOPKiAIwq740Mse2H79rGvnYEnb9UoreOVRaMEmYv test@pi
ssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAACAQCm1+pkpakcPp10Ql7TzTnvdhzenq5nuFa8MvOD/WeQPa36pJZQWeppkPZaxWfQzSmr8+/16nYgt1uWLiD/rlYsq5gNpGfA8Xk7Z6aUU6zfG8RDAhd9d8wgVCzeXrYj8z9FtzhdFTUfJZQkbgUu3kxAIXu0c66LsJ9C1fHI2bDRcVhYMk3ugFJWrpG0FPSOhE7CSwNHzeQcnPscjUq4w40nke90RnHtt4OgHD7dMW+clUT19kE7Jg9GSkbNrcTSH0JudXHApC49QfNnaKEn58y4hL6DnDViveNOu6ajki5JVUJAG550tw4NNBGZzq66masuHI3Jq4iTYzRYf802SiS/ylxwEWClxwmkor5MrhbVgMA8Jt3ijOdbDFs5DD6QGoIfFj2RsEtPxg8UXu6bYn2I14GlBIGpxmg7a+Gtsl0hgqMgPEbwjregOf7MqvQ6fg5pQHlMSxrNdqs5mfF2xqD7QptBdOBKaVDw9GJZaoO4ehEUSeAJIARBf+59THjvVm1zA/a7ZA2i+wb/fPCGvdSjb9M/4X/c8k0ySGym/wtwap+9lYUMaKU4zKsfx2Kjv57g5i5cEFneRbInZXaTaG8gxRyPend+VaLFcHOyL1hXoxApri5wVELZCjUxdlk9QNc5gUdKL/8c5TfzrfMjMly8Rlvd8GdATD8r9uoWglPtWw== test2@piadmin@router:/tmp/home/root#
This is a line in authorized_keys... webguidontcare
</pre>

tuxpeople, 2018-07-10:
@nullcake thanks for checking. Let's say it's my fault an close this issue.

closed 2018-07-10 by tuxpeople

thowi, 2019-07-06:
I'm actually not using Merlin (vanilla RT-AC86U firmware 3.0.0.4.384_45713), but I'm seeing a similar problem, so maybe it's related.

I can add a second key, save, and all will work fine.
However, once the router is rebooted, the second key is gone, and only the first one is left.

Not sure how this is actually saved between reboots (NVRAM?), but maybe there's some character limit.

aselvan, 2020-04-19:
> I'm actually not using Merlin (vanilla RT-AC86U firmware 3.0.0.4.384_45713), but I'm seeing a similar problem, so maybe it's related.
>
> I can add a second key, save, and all will work fine.
> However, [...]

I have the same issue of keys disappearing (except the first one) after reboot. Is there any workaround? I am not on Merlin but on Asus stock firmware (Version:3.0.0.4.384_8011) on AX-11000 but the problem is same. 

I tried to force saving on a ssh login like so below before reboot but still the 2 keys out of the 3 gets wiped after reboot. Note: the last command does show all 3 keys are in the nvram!

```
/tmp/home/root# nvram save
/tmp/home/root# nvram commit
/tmp/home/root# nvram show |grep ssh-rsa
```