RMerl/asuswrt-merlin.ng #158: VLAN documentation

TGM, 2018-06-02:
Hello,

Can you integrate VLAN support into the GUI?
Is there any complete documentation available for CLI?

Thank you

Geezus42, 2018-06-02:
It seems like the link aggregation documentation is fairly out of date as well. At the very least it is missing some important details because it does not work. AC-5300

RMerl, 2018-06-02:
There's no documentation, as it requires access to Broadcom's proprietary documentation.

There are no plans to implement VLAN support, due to lack of documentation and also conflict with Asus's own implementation.

closed 2018-06-02 by RMerl

samip5, 2019-07-28:
VLAN's are currently not supported at all on the device, even if it's just in AP mode?

aenertia, 2020-04-03:
vlan's work - the problem seems to be in whatever customized services deal with bridges on the devices:
i.e if i:
ip link add link eth3 name eth3.10 type vlan id 10 
ip link set up eth3.10
ip a a 192.168.2.1/24 dev eth3.10
ping 192.168.2.2 (to a client connected over eth3 with tagged interface on vlan10)
---
Everything works fine.

The problem seems to be related to bridges

If I then do :
ip a f dev eth3.10
ip link set down eth3.10
ip link add name newbr0 type bridge
ip link set eth3.10 master newbr0
---

Everything looks good so far, the second I
ip link set up dev newbr0
---

It will trigger a hard crash on the router, requiring a power-cycle.

This is massively problematic as it means you can fence your Internet of Shit devices into an isolated segment on wire. And means that WPA2-Enterprise Dyanmic VLAN assignment ; including physically connected devices is broken.

samip5, 2020-04-03:
> Vlans Work - Bridges do not.
> 
> vlan's work - the problem seems to be in whatever customized services deal with bridges on the devices:
> i.e if i:
> ip link add link eth3 name eth3.10 type vlan id 10
> ip link set up eth3.10
> ip a a 192.168.2.1/24 dev eth3.10
> ping 192.168.2.2 (to a client connected over eth3 with tagged interface on vlan10)
> 
> Everything works fine.
> 
> The problem seems to be related to bridges
> 
> If I then do :
> ip a f dev eth3.10
> ip link set down eth3.10
> ip link add name newbr0 type bridge
> ip link set eth3.10 master newbr0
> 
> Everything looks good so far, the second I
> ip link set up dev newbr0
> 
> It will trigger a hard crash on the router, requiring a power-cycle.
> 
> This is massively problematic as it means you can not fence your Internet of Shit devices into an isolated segment on wire. And means that WPA2-Enterprise Dyanmic VLAN assignment ; including physically connected devices is broken.

Do you happen to have a SERIAL connection to the router by any chance? I think it would be beneficial to know what actually happens when it crashes which should be visible only from the on-board serial if there is one.