RMerl/asuswrt-merlin.ng #162: mDNS / Bonjour not forwarding WLAN -> LAN

Speedliner, 2018-06-09:
I have the same issue as discribed in #102. I am not able to see any devices via bonjour on my wired network part with my wireless devices and vice verca (e.g. iPhone to Apple TV; iPhone to Synology NAS).

Tried severla wireless settings under the "Professional" tab - but the issue persists.

RT-AC86U, v384.5 

Regards

the0ne, 2018-09-29:
@Speedliner I guess https://github.com/RMerl/asuswrt-merlin.ng/issues/154 does describe what you experience, right?

Speedliner, 2018-10-04:
> 
> 
> @Speedliner I guess #154 does describe what you experience, right?

Not really. If I get #154 right, communication is not possible between WLAN devices at all. 

I'm facing a different issue concerning bonjour / mDNS and Apple devices. 

But after resetting the configuration of my RT-AC86U, I have a slightly different issue:

After powering on certain Apple / Bonjour devices, they are able to see each other for a short time (10 to 30 seconds). But after that, the bonjour / mDNS connection gets lost.

fusionstream, 2018-10-14:
In #206, it seemed to have resolved itself after an accidental (or buggy) loss of my configuration, whereupon I restored from a backup and it's been working for more than 12 hours now.

closed 2019-01-15 by themiron