RMerl/asuswrt-merlin.ng #163: RT-AC87U CPU Core 2 Always at 100%

MichaelJAllen, 2018-06-10:
Version 384.5

Despite rebooting and re-flashing, the web-interface reports that CPU Core 2 is always at 100% utilization. Seems like something is infinitely looping (if there's anything I can do to figure out what that might be, let me know).

tuapuikia, 2018-06-18:
May I know what version of firmware you are using?

MichaelJAllen, 2018-06-21:
Sure. It's right there on the first line of my original report. ;-)

tuapuikia, 2018-06-22:
May I know what features is enabled? AiProtect? DNS filter? File server? and etc.

bdlou, 2018-07-23:
Wanted to add a +1 here.  Seeing the same thing.  Was seeing this before I enabled any of the protection stuff. Seeing it afterwards as well. 384.5 as well.  Let me know if there's any logs I can collect or anything.

brunnels, 2018-07-29:
I was experiencing the same thing on RT-AC88U, same version.  Soft reboot several times didn't resolve anything.  I did a hard power cycle and the issue resolved.

Atera, 2018-08-02:
Seeing the same issue. Recently started using QoS, but after seeing the 100 % utilisation (and the resulting routing slowdown) I tried disabling it again to see if it would help. It has not helped. Having to hard restart every few days to resolve the issue whenever it arises.

themiron, 2018-08-02:
what process is causing this?
check with http://man7.org/linux/man-pages/man1/top.1.html via telnet/ssh.

joehillen, 2018-08-21:
I ssh'd into the router when I noticed one of the cores was at 100%. Everything looks normal in top. No processes were using more than 20%.

alexejk, 2018-09-09:
Having the same issuue from time to time with always latest firmware running on RT-AC88U. When temp closes 80C mark on CPU - throughput starts to drop. While no processes seem to take more than 1%CPU it would seem IO is taking almost 50%...

Generally a hard reboot of router solves this for some time.

ldrrp, 2018-09-10:
i had to factory reset my router to get this working.

Atera, 2018-10-23:
> 
> 
> what process is causing this?
> check with http://man7.org/linux/man-pages/man1/top.1.html via telnet/ssh.

As joehillen has mentioned, when I ssh in, nothing seems out of the ordinary. No process shows up as taking up more than 2% CPU usage, and adding up the percentages that appear on top's output would make it seem that less than 8% of CPU 2's capacity was being used. And yet on the main overview page CPU usage for CPU 2 is locked in at 100%. The top processes are nt_monitor, /etc/openvpn/vpnclient1, /sbin/netool, httpd -i br0, protect_srv, /sbin/wanduck and mastiff.

Mem 264252K used, 250932K free, 3828K shrd, 932K buff, 13640K cached
CPU0 0.0% usr 20.0% sys, 0.0% nic 80.0% idle, 0.0% io 0.0% irq 0.0% sirq
CPU0 0.0% usr 20.0% sys, 0.0% nic 0.0% idle, 80.0% io 0.0% irq 0.0% sirq

Load average 6.04 6.08 6.08 1/119 23684

olskar, 2018-11-08:
I have the same issue but with CPU 1, top doesnt show any process that use that much cpu..

shivaprasadh, 2018-11-17:
+1 on this issue.  Very often I too see this issue on RT-AC68U.  Core 2 is always on 100% from the Web status page.

yozik04, 2018-12-18:
+1 on RT-AC68U. 384.7_2
Load average is high 6.17 6.14 6.14 1/118 24334

Can install opkg packages and debug. But have no idea what to look for.

yozik04, 2018-12-18:
And most probably the reason is this command:
`cp /tmp/syslog.log /tmp/syslog.log-1 /jffs`

Please try next command if you see problems.
`# ps | grep jffs`
Do you have **_cp_** command hanging there?

alexejk, 2018-12-30:
It seems that ~50% is taken by io ?
Seem to be the case with both my output (below) and what some others linked

```
Mem: 235892K used, 279292K free, 3192K shrd, 3876K buff, 17488K cached
CPU:  0.5% usr  1.3% sys  0.0% nic 48.8% idle 47.9% io  0.0% irq  1.2% sirq
Load average: 5.02 4.65 4.30 2/141 4130
```

yozik04, 2018-12-31:
Please check if /jffs is readable for you.

alexejk, 2018-12-31:
> Please check if /jffs is readable for you.

If that was for my comment then yes it is. The 50% IO & 100% CPU on one core (which is what 50%IO in ´top´ actually means)  happens only after some time after reboot. I think it took 2-3 weeks before it appeared again.

Migelo, 2019-01-02:
I have the same issue issue with the RT-AC56U 380.59-0. Over time the load creeps up and up, reaching over 6.0 currently. I am unable to open the web interface or establish a VPN connection with OpenVPN. 
/jffs is writeable, I unmounted the USB drive connected to the router.

`Mem: 113432K used, 142268K free, 0K shrd, 1004K buff, 9416K cached
CPU:  1.3% usr  8.8% sys  0.0% nic 16.8% idle 39.7% io  0.0% irq 33.1% sirq
Load average: 6.40 6.29 6.01 2/111 3441`

`$ ps
  PID USER       VSZ STAT COMMAND
    1 nadzorni  6428 S    /sbin/preinit
    2 nadzorni     0 SW   [kthreadd]
    3 nadzorni     0 SW   [ksoftirqd/0]
    4 nadzorni     0 SW   [kworker/0:0]
    5 nadzorni     0 SW   [kworker/u:0]
    6 nadzorni     0 SW   [migration/0]
    7 nadzorni     0 SW   [migration/1]
    9 nadzorni     0 SW   [ksoftirqd/1]
   10 nadzorni     0 SW<  [khelper]
   11 nadzorni     0 SW   [kworker/u:1]
   57 nadzorni     0 DW   [sync_supers]
   59 nadzorni     0 SW   [bdi-default]
   60 nadzorni     0 SW<  [kblockd]
  113 nadzorni     0 SW   [kswapd0]
  159 nadzorni     0 SW   [fsnotify_mark]
  167 nadzorni     0 SW<  [crypto]
  239 nadzorni     0 SW   [mtdblock0]
  244 nadzorni     0 SW   [mtdblock1]
  249 nadzorni     0 SW   [mtdblock2]
  254 nadzorni     0 SW   [mtdblock3]
  280 nadzorni     0 SW   [kworker/0:1]
  281 nadzorni     0 SW   [kworker/1:1]
  284 nadzorni     0 SW   [mtdblock4]
  289 nadzorni     0 SW   [mtdblock5]
  293 nadzorni   664 S    hotplug2 --persistent --no-coldplug
  330 nadzorni  6268 S    console
  333 nadzorni  1388 S    /bin/sh
  341 nadzorni  1380 S    /sbin/syslogd -m 0 -S -O /tmp/syslog.log -s 256 -l 7
  343 nadzorni  1384 S    /sbin/klogd -c 5
  411 nadzorni  6272 S    /sbin/wanduck
  413 nadzorni  1076 S    dropbear -p 10.0.0.1:22 -a -j -k
  415 nadzorni  1176 S    /bin/eapd
  417 nadzorni  1776 S    /bin/wps_monitor
  418 nadzorni  6272 S    wpsaide
  420 nadzorni  1736 S    nas
  423 nadzorni  1364 S    /usr/sbin/acsd
  432 nadzorni  6572 D    httpd -i br0
  434 nadzorni  1392 S    crond
  435 nadzorni  1148 S    /usr/sbin/infosvr br0
  442 nadzorni  6272 S    ots
  443 nadzorni  2276 S    rstats
  445 nadzorni  1144 S    cstats
  450 nadzorni  1192 S    lld2d br0
  454 nadzorni 41676 D    networkmap --bootwait
  505 nadzorni  6272 S    bwdpi_check
  565 nadzorni  1388 S    /sbin/udhcpc -i eth0 -p /var/run/udhcpc0.pid -s /tmp/udhcpc -O33 -O249
  584 nadzorni  1388 S    /usr/sbin/pppd file /tmp/ppp/options.wan0
  591 nadzorni     0 SW   [khubd]
  789 nadzorni  6272 S    usbled
  790 nadzorni  2200 S    u2ec
  791 nadzorni  1220 S    lpd
  802 nadzorni  2200 S    u2ec
  803 nadzorni  2200 S    u2ec
  936 nadzorni  2220 S    avahi-daemon: running [RT-AC56U-E9B0.local]
  951 nadzorni  5476 S    nmbd -D -s /etc/smb.conf
  952 nadzorni  5328 S    nmbd -D -s /etc/smb.conf
  953 nadzorni  5548 S    /usr/sbin/smbd -D -s /etc/smb.conf
  958 nadzorni     0 SW<  [rpciod]
  963 nadzorni  1380 S    /sbin/zcip -p /var/run/zcip0.pid eth0 /tmp/zcip
  966 nobody     672 S    /usr/sbin/portmap
  968 nadzorni   836 S    /usr/sbin/statd
  971 nadzorni     0 SW   [lockd]
  972 nadzorni     0 SW   [nfsd]
  974 nadzorni   904 S    /usr/sbin/mountd
 1020 nadzorni  6272 S    ntp
 1134 nadzorni  1120 S    dropbear -p 10.0.0.1:22 -a -j -k
 1135 nadzorni  1396 S    -sh
 1253 nadzorni   860 S    miniupnpd -f /etc/upnp/config
 1255 nadzorni   664 S    pptpd -c /tmp/pptpd/pptpd.conf -o /tmp/pptpd/options.pptpd
 1256 nadzorni   696 S    /usr/sbin/bcrelay -i br0 -o ppp1[0-9].* -n
 1303 nadzorni  6404 D    disk_monitor
 1357 nadzorni  2200 S    wred -B
 1360 nadzorni  2200 S    wred -B
 1361 nadzorni  2200 S    wred -B
 1362 nadzorni  2200 S    wred -B
 1363 nadzorni  2200 S    wred -B
 1364 nadzorni  2200 S    wred -B
 1365 nadzorni  2200 S    wred -B
 1366 nadzorni  2200 S    wred -B
 1367 nadzorni  2200 S    wred -B
 1368 nadzorni  2200 S    wred -B
 1369 nadzorni  2200 S    wred -B
 1397 nadzorni  6272 S    bwdpi_wred_alive
 1402 nadzorni  4048 S    data_colld -i 1800 -p 43200 -b -w /tmp/bwdpi/dc/
 1403 nadzorni  4048 S    data_colld -i 1800 -p 43200 -b -w /tmp/bwdpi/dc/
 1404 nadzorni  4048 S    data_colld -i 1800 -p 43200 -b -w /tmp/bwdpi/dc/
 1405 nadzorni  4048 S    data_colld -i 1800 -p 43200 -b -w /tmp/bwdpi/dc/
 1406 nadzorni  4048 S    data_colld -i 1800 -p 43200 -b -w /tmp/bwdpi/dc/
 1407 nadzorni  4048 S    data_colld -i 1800 -p 43200 -b -w /tmp/bwdpi/dc/
 1426 nadzorni  4048 S    data_colld -i 1800 -p 43200 -b -w /tmp/bwdpi/dc/
 1452 nadzorni  6272 S    hour_monitor
 1632 nadzorni  6272 S    hotplug block
 1633 nadzorni  6272 D    hotplug block
 2541 nadzorni  1120 S    dropbear -p 10.0.0.1:22 -a -j -k
 2544 nadzorni  1392 S    -sh
 2657 nadzorni  1120 S    dropbear -p 10.0.0.1:22 -a -j -k
 2658 nadzorni  1404 S    -sh
 3527 nadzorni  1120 S    dropbear -p 10.0.0.1:22 -a -j -k
 3528 nadzorni  1392 S    -sh
 5770 nadzorni  1384 R    ps
10857 nadzorni  1384 S    sh -c bwdpi_sqlite -e -s NULL
10858 nadzorni  5132 D    bwdpi_sqlite -e -s NULL
16549 nadzorni   836 S    /usr/sbin/statd
16556 nadzorni   904 S    /usr/sbin/mountd
16567 nadzorni  5328 S    nmbd -D -s /etc/smb.conf
24450 nadzorni   836 S    /usr/sbin/statd
24455 nobody    1332 S    dnsmasq --log-async
24457 nadzorni   904 S    /usr/sbin/mountd
24468 nadzorni  5328 S    nmbd -D -s /etc/smb.conf
27428 nadzorni     0 SW   [kworker/1:2]`

Migelo, 2019-01-29:
It would be interesting to see whether killing the process trying to read/write the `/jffs` partition would fix this high load, I guess it would. Beside grepping for any mention of _jffs_ could someone suggest any other way of figuring out what is reading/writing to the partition with the limited tool set available?

yozik04, 2019-01-29:
In my situation it was
`cp /tmp/syslog.log /tmp/syslog.log-1 /jffs`
And it was not killable even with **-9**

Migelo, 2019-01-29:
Interesting. I set my log settings to only log `CRITICAL` events and let me see if it takes it longer for the router to freeze.

neoKushan, 2019-02-18:
Has anyone figured out anything more on this? I'm suffering an issue on my RT-AX88U with firmware 384.9. When I log into the Web UI, it's showing all 4 CPU's being pegged almost constantly:

[image]

If I ssh in and run top, it doesn't show anything beyond bcmsw_rx gobbling up CPU:

[image]

This is causing my connection to suffer high pings and dropouts, until I reboot the router and it's all fine for a few hours:

[image]

I'm not too familiar with Linux so I'm not really sure where to look beyond this. Anyone got any suggestions?

neoKushan, 2019-03-26:
So just as an update in case anyone finds this via Google.

While on 384.9, I factory reset the router and it made no difference. I gave in and set a scheduled daily reboot to make it bearable but it would still crop up at night.

The good news is that I tried the last beta of 384.10 4 or 5 days ago and it has been stable since. I've even disabled the daily reboot and not seen any issues. I assume whatever it was that caused this for me has been fixed in 384.10. I'll be trying out the stable 384.10 later tonight.

ldrrp, 2019-03-29:
Not sure if its related but i lost internet today (Thanks spectrum) and i noticed the cpu spiked to 100% becouse the router was trying to get time for ntp every second according to the logs.

Mensra, 2019-03-29:
Whenever I am transferring via SMB (samba) my sirq pegs the needle on cpu 0, and RAM usage heads for the sky with close to 100% as well. When transfer completes, all is well. Transfer speed are about 4 Mbps. Other then that, numbers are normal. No one process seems to be a CPU hog. Running 384.10_beta3, but had the same issue on 384.9 Running on RT-AC3100.

RMerl, 2019-03-29:
Guys, please use the support forums at SNBForums if you are in need of support.  High CPU usage is not a bug, it`s simply a sign that something is running and using the CPU.

High CPU usage while accessing a network share for instance is perfectly normal - these routers have low-powered CPUs that are not optimized for SMB/USB file sharing.

closed 2019-03-29 by RMerl

diegocr, 2019-03-30:
> low-powered CPUs that are not optimized for SMB/USB file sharing

Huh?! That statement is quite unfortunate. If SMB - or any other component - is not optimized for the CPU and hence could cause problems why is it being included with the firmware in the first place?

I do own a Commodore Amiga with a 60Mhz CPU where SMB does work perfectly on it...

RMerl, 2019-03-30:
You're not gonna get 1 GBps of SMB throughput out of that Amiga with zero CPU usage...  This is what some people are expecting out of their router.

Migelo, 2019-03-30:
In my case I wasn't using any SMB sharing and the CPU usage would at some point just get stuck at 100%.

neoKushan, 2019-03-30:
I'd just like to wade in and say that my issue was also that the CPU was
being pegged at 100% without anything running (see my screenshots above).

The attitude being displayed here is very disappointing - of course we know
that activity will cause CPU usage and of course we know that these devices
are low power, but that's not the problem - the problem is quite clearly an
error of some kind. Dismissing people due to obviously not reading their
responses is poor form to say the least.Q

Atera, 2019-03-31:
I agree with neoKushan. In my case as well, no shares or anything were taking place. Just a handfull of machines connected to a router with one of them configured to use a VPN. CPU usage just gets stuck at 100% every couple weeks and requires a hard restart (soft restart will not help). And no, usage is not dependant on load or VPN usage. SSHing in showed that it was all IO taking up all the resources, and after a while no packets can get in or out and the router management itself becomes unresponsive. And most of the times when this happens, only moderate load is taking place, with the device configured to use VPN not connected/turned on.

mbze430, 2019-04-14:
I am also having this issue with my ASUS RT-AC5300.  What I have found, at least in my case, is that if I turn off Adaptive QoS the CPUs goes back to normal.  Turning Adaptive QoS back on starts to eat up CPU for me.  Regardless if I reboot or hard reset.  As long as Adaptive QoS is enabled it goes crazy.  Traditional QoS is fine.

I am on firmware 384.10_2

cshumac, 2019-04-28:
I have this same issue, Asus RT-AC3100 on Firmware Version:384.10_2. Why is this issue closed when its still an issue?

Originalimoc, 2019-07-07:
Some clients can be "malicious" to cause this problem. [bcmsw_rx] process means CPU is processing network traffic. Devices such as Google Home/Chromecast may keep broadcasting for no good reason and since NAT acceleration can break network monitoring utility(iftop/tcpdump) and there's no way to turn this feature off on some models it's hard to diagnose exactly which device is the cause. Try `service restart_wireless` when the problem arises to see if there's any good.

djbooya, 2019-09-23:
> And most probably the reason is this command:
> `cp /tmp/syslog.log /tmp/syslog.log-1 /jffs`
> 
> Please try next command if you see problems.
> `# ps | grep jffs`
> Do you have **_cp_** command hanging there?
> 
> cp command is not killable... Even with -9.

I am still seeing this issue with 384.13 on a RT-AC66U_B1.

This is what my load looks like when it's in the bad state:
[image]

The only way to fix is rebooting since kill -9 doesn't kill the hung process.  Anyone have any other solutions or data I should try to gather?  I am running 5 other devices with Merlin on them and this is the only one that exhibits the issue.

Migelo, 2019-09-27:
I found a [thread](https://www.snbforums.com/threads/syslog-on-jffs.13899/) started in 2013 where people have discovered ASUS implemented this copying of the `syslog.log` and `syslog.log-1` to `/jffs`. 

So I followed the advice in that thread and put this into my `/jffs/scripts/services-start`. Make sure you make it executable with `chmod +x /jffs/scripts/services-start`.

```
#!/bin/sh

# kill syslogd and remove files
/usr/bin/killall syslogd
rm -f /jffs/syslog.log
rm -f /jffs/traffic.db

# Remount /jffs partition readonly
mount -o remount,ro,noatime /jffs

# start syslogd, see `syslogd -h` for more details
syslogd -m 0 -S -O /tmp/syslog.log -s 256 -l 7

```

djbooya, 2019-09-27:
I'll give it a try..i guess I can always re-mount /jffs as rw when I need to write to it..and then move it back to ro when done...

middleagedman, 2019-10-09:
Same issue here.. rectified somewhat by making /jffs/syslog.log and /jffs/syslog.log-1 directories.  Now at least syslog doesn't try to rotate or write

djbooya, 2019-10-09:
The RO filesystem has appeared to be stable the past 2 weeks for me.  Looks promising..  not sure what makes this 1 router different from the others I have.

lobocobra, 2020-03-23:
I use Merlin 384.15 on a RT-AC88u

in my case the 2 CPU switch from 0% to 100% (and simultaniously vice versa) every second.
=> seems to be a similar behavior as described above, but for sure not normal

[image]

jrowe88, 2020-09-11:
> Has anyone figured out anything more on this? I'm suffering an issue on my RT-AX88U with firmware 384.9. When I log into the Web UI, it's showing all 4 CPU's being pegged almost constantly: [...]

I had a similar problem and just figured it out.  My NAS has two gigabit NICs "bonded" using balance-rr (round robin).  Both were plugged into the AX88U switch and any time files were copied CPU would go crazy.  I am sure it was unnecessarily causing the switch to do a lot of work with the packets.  I unplugged one cable and it instantly started working normally.  I'm using balance-xor (which balances by MAC address, not packet) and all it good.

Originalimoc, 2020-09-11:
``service stop_nasapps``, as name suggests.