RMerl/asuswrt-merlin.ng #164: Support of RT-AC66U

kri5, 2018-06-12:
It seems that the latest firmware available for the RT-AC66U in dating from the 21/05/2018 is at version 3.0.0.4.382.50470, so using branch 382. (https://www.asus.com/Networking/RTAC66U/HelpDesk_BIOS/)

Does this help in supporting it in the ng branch of merlin?

RMerl, 2018-06-12:
Support for the RT-AC66U and RT-N66U has been dropped.

closed 2018-06-12 by RMerl

kri5, 2018-06-12:
Any reasons why theses models support has been dropped?

dlehammer, 2018-09-08:
> Support for the RT-AC66U and RT-N66U has been dropped.

@RMerl source of statement needed.

I just checked

Diaoul, 2018-10-30:
I was looking for a reason why the support has been dropped and a quick search lead me here. I wish this release from Asus allowed for continued support of the Merlin firmware.

RMerl, 2018-10-30:
Explanations were posted on SNBForums at the time, but to sum them up here (for future reference):

- Too many different models to support for one single developer, so some had to be dropped when the work began on the NG code branch
- RT-N66U and RT-AC66U are the only two devices still based on the older MIPS architecture and an older kernel, requiring a lot of dedicated work specificaly for these two devices versus all the newer ones
- Infrequent GPL releases by Asus, meaning that half of the time it would be impossible for me to compile a working build due to lack of compatible binary blobs
- Technical limitations of this older platform (older kernel, requires different and older versions of iptables, ipset, iproute, etc...)

th333boo, 2021-01-19:
Thanks for your answer @RMerl