RMerl/asuswrt-merlin.ng #165: Add user-defined options (77 & 90) for WAN DHCP requests (Orange French ISP)

Ahbrasax, 2018-06-16:
Hello,

The present release branch 384.5 support two WAN DHCP options : 60 and 61.

My current Internet access is provided by Orange ISP in France. This one allows me two methods of wan connexion : 

An old connection type based on PPPoE protocol and  an other based on dynamic IP attribution (IPoE). 

At present, Orange ISP is trending to migrate all of their customer to IPoE for the new xDSL and fiber customers.

Anyway, if I want to get an IP though DHCP on my WAN connection, I need to provide the following DHCP options :

- **(60)** vendor-class-identifier=sagem
- **(77)** userclass=+FSVDSL_livebox.Internet.softathome.Livebox3
- **(90)** authsend=fti/xxxxxxx  ## xxx hidden for anonymity, it is my ISP personal identifier

This is required by my ISP for identification purpose.
All these values need to be supplied in hex format

**Linux dhclient example :** 

```
send vendor-class-identifier = "sagem";
option userclass code 77 = string;
send userclass 2b:46:53:56:44:53:4c:5f:6c:69:76:65:62:6f:78:2e:49:6e:74:65:72:6e:65:74:2e:73:6f:66:74:61:74:68:6f:6d:65:2e:4c:69:76:65:62:6f:78:33;
option authsend code 90 = string;
send authsend 00:00:00:00:00:00:00:00:00:00:00:66:74:69:2f:*:*:*:*:*:*:*; ## replace star with your identifier in hex format..
```

If this DHCP options code were implemented udhcpc command line should be : 

```
/sbin/udhcpc -i br1 -p /var/run/udhcpc0.pid -s /tmp/udhcpc -O33 -O249  -x 77 2b46535644534c5f6c697665626f782e496e7465726e65742e736f66746174686f6d652e4c697665626f7833 -x 90 00000000000000000000006674692fxxxxxxxxxxxxxx -V sagem
```

**DHCP dump WAN request (Wireshark)**

```
802.1Q Virtual LAN, PRI: 6, CFI: 0, ID: 832
Bootstrap Protocol (Discover)
    Message type: Boot Request (1)
    Hardware type: Ethernet (0x01)
    Hardware address length: 6
    Hops: 0
    Transaction ID: 0x0f48395c
    Seconds elapsed: 1
    Bootp flags: 0x8000, Broadcast flag (Broadcast)
        1... .... .... .... = Broadcast flag: Broadcast
        .000 0000 0000 0000 = Reserved flags: 0x0000
    Client IP address: 0.0.0.0
    Your (client) IP address: 0.0.0.0
    Next server IP address: 0.0.0.0
    Relay agent IP address: 0.0.0.0
    Client MAC address: Sagemcom_xx:yy:zz (2c:39:96:xx:yy:zz)
    Client hardware address padding: 00000000000000000000
    Server host name not given
    Boot file name not given
    Magic cookie: DHCP
    Option: (53) DHCP Message Type (Discover)
        Length: 1
        DHCP: Discover (1)
    Option: (55) Parameter Request List
        Length: 11
        Parameter Request List Item: (1) Subnet Mask
        Parameter Request List Item: (3) Router
        Parameter Request List Item: (6) Domain Name Server
        Parameter Request List Item: (15) Domain Name
        Parameter Request List Item: (28) Broadcast Address
        Parameter Request List Item: (51) IP Address Lease Time
        Parameter Request List Item: (58) Renewal Time Value
        Parameter Request List Item: (59) Rebinding Time Value
        Parameter Request List Item: (90) Authentication
        Parameter Request List Item: (119) Domain Search
        Parameter Request List Item: (120) SIP Servers
    Option: (60) Vendor class identifier
        Length: 5
        Vendor class identifier: sagem
    Option: (77) User Class Information
        Length: 44
        Instance of User Class: [0]
            User Class Length: 43
            User Class Data: 46535644534c5f6c697665626f782e496e7465726e65742e736f66746174686f6d652e4c697665626f7833
    Option: (90) Authentication
        Length: 22
        Protocol: configuration token (0)
        Algorithm: 0
        Replay Detection Method: Monotonically-increasing counter (0)
        RDM Replay Detection Value: 0x0000000000000000
        Authentication Information: fti/*******
    Option: (255) End
        Option End: 255

```
Could you please expose option 77 and 90 in the asuswrt-merlin Gui ?

Regards

crkinard, 2018-06-16:
Use the scripting system provided to add the lines to dnsmasq on startup.

Ahbrasax, 2018-06-17:
Well, my request concern Dhcp client instanciated when wan interface goes up.
When using IPoE protocol public IP is retrieved through Dhcp client software (/sbin/udhcpc).

Adding conf  lines to dnsmasq on startup ? 
Well in my mind, dnsmasq concerns  Dhcp Server functionality dedicaded to Lan clients.

Am I wrong ?

RMerl, 2018-07-06:
I don't want to add any further ISP-specific setting because it would be a never ending chase to keep up-to-date.  Options 60 and 61 were partly supported by the stock firmware (for DSL users), I simply extended it to regular DHCP users.

closed 2018-07-06 by RMerl

dexter74, 2019-02-12:
Can you write tutorial please.

jeyf, 2021-05-04:
Hello,
the optical fiber arrives this year everywhere in France !! and dhcp option 90 is required .. otherwise most asus routers will go into the trash. there are going to be millions of users who will need it. I think this modification is important and only concerns a few isolated cases. if you could reconsider it that would be really great.