RMerl/asuswrt-merlin.ng #166: VPN Server - OpenVPN - Instance 1: internet only malfunction.

janusn, 2018-06-20:
### Synopsis:
The client connect to the server cannot access to internet nor LAN when setting the option of "VPN Server / OpenVPN / Server 1 / Client will use VPN to access" to "Internet only". On the other hand, the "Internet only" option works as intended on Server 2 instance.

### Steps to reproduce:

1. Login into the webUI of the asuswrt-merlin.
2. Click on the "VPN" tab on the left
3. Click on "VPN Server" tab on the top.
4. Select "Server 2" from "Server instance"
5. Select "Internet only" from "Client will use VPN to access"
6. Create an account on the bottom.
7. Click "Apply" button.
8. Export and import the profile into an iOS client running OpenVPN Connect.
9. Connect to the server with the profile exported and corresponding credential on the iOS client.
10. Browse to "https://www.whatismyip.com/" in Safari on the iOS client.

### Expected result:

The webpage loaded and shows the IP of the router.

### Actual result:

The page cannot be loaded.

### Configurations:
- firmware version: 384.5
- Router: ASUS RT-AC68U
- **client:**
  - OS: iOS 11.4(15F79)
  - Hardware: iPhone X
  - client software: OpenVPN Connect version 1.2.9

RMerl, 2018-07-06:
Unable to reproduce, working for me.  Compare the settings of the two server instances,

RMerl, 2018-07-06:
One thing to note, if you set it to Internet-only, you must ensure the DNS server used by the client is a public one, and not your router's IP.,

janusn, 2018-07-07:
Thanks for the feedback. I have repeated back and forth configured the servers several times between "Both" and "Internet only". The configuration were done through the WebUI with a Mac connected to the router through WiFi and the resulting VPN servers were tested on an iPhone connected to cellular network only. The internet service via OpenVPN on instance server 1 worked only when set to "Both". On the other hand, the instance Server 2 worked with either "Internet only" or "Both".

As a few people depended on the OpenVPN for internet access already, I had to copy all the settings on instance Server 1 to instance Server 2 and reconfigured the instance Server 1 for a new usage.

I have just tested "Internet only" on instance Server 1 again. It didn't work.

closed 2020-09-21 by RMerl