RMerl/asuswrt-merlin.ng #168: Guest wifi user can access subnets connected through OpenVPN

MichaelEFlip, 2018-06-24:
Users in guest wifi can access subnets which are connected through OpenVPN.
Access for guest users to devices in local subnet is forbidden.

Whish: No routing into VPN tunnels for guest wifi users

RT-AC86U
Firmware Version: 384.5

This issue was already present in the legacy branch (issue tracker closed).

locksmithone, 2018-06-28:
Does the guest user have the VPN tunnel password? If the guest user has legitimate access to the VPN tunnel, shouldn't they have access to the subnets provided by the VPN tunnel, regardless of where the user is? (In this case, it happens that the user is connected to the router own guest network, but what if this user was connecting from across the globe?)

RMerl, 2018-08-17:
Technically, a tunnel isn't part of an Intranet, therefore this is working as intended.  A lot of users actually use these VPN tunnels to redirect their Internet traffic, in which case they'll need their Guest clients to also route through it.  If you need finer control, you will have to manually configure firewall rules to achieve it.

closed 2018-08-17 by RMerl