RMerl/asuswrt-merlin.ng #170: iptables appears to be missing extension libraries

nullcake, 2018-07-07:
Version:
ASUSWRT-Merlin RT-AC68U 384.5-0 Sun May 13 01:53:17 UTC 2018

Web History appears to not work due to a missing iptables library for webmon. Other iptables extensions may also be missing.

See:
http://merlin-ip/AdaptiveQoS_WebHistory.asp

Ex: iptables webmon lib appears to be missing

<pre>
admin@router:/tmp/home/root# cd /
admin@router:/# find . -name 'libipt*'
./usr/lib/libiptc.so
./usr/lib/libiptc.so.0
./usr/lib/libiptc.so.0.0.0
./usr/lib/xtables/libipt_CLUSTERIP.so
./usr/lib/xtables/libipt_DNAT.so
./usr/lib/xtables/libipt_ECN.so
./usr/lib/xtables/libipt_LOG.so
./usr/lib/xtables/libipt_MASQUERADE.so
./usr/lib/xtables/libipt_MIRROR.so
./usr/lib/xtables/libipt_NETMAP.so
./usr/lib/xtables/libipt_REDIRECT.so
./usr/lib/xtables/libipt_REJECT.so
./usr/lib/xtables/libipt_ROUTE.so
./usr/lib/xtables/libipt_SAME.so
./usr/lib/xtables/libipt_SNAT.so
./usr/lib/xtables/libipt_TOS.so
./usr/lib/xtables/libipt_TRIGGER.so
./usr/lib/xtables/libipt_TTL.so
./usr/lib/xtables/libipt_ULOG.so
./usr/lib/xtables/libipt_account.so
./usr/lib/xtables/libipt_ah.so
./usr/lib/xtables/libipt_geoip.so
./usr/lib/xtables/libipt_icmp.so
./usr/lib/xtables/libipt_ipp2p.so
./usr/lib/xtables/libipt_layer7.so
./usr/lib/xtables/libipt_realm.so
./usr/lib/xtables/libipt_tos.so
./usr/lib/xtables/libipt_ttl.so
./usr/lib/xtables/libipt_unclean.so
./usr/lib/xtables/libipt_web.so
./usr/lib/xtables/libipt_webstr.so
admin@router:/# iptables -A web_monitor -p tcp -m webmon --max_domains 2000 --max_searches 2000
iptables v1.4.15: Couldn't load match `webmon':No such file or directory

Try `iptables -h' or 'iptables --help' for more information.
</pre>

RMerl, 2018-07-07:
WebHistory is based on the Trend Micro DPI engine, it does not use iptables.

closed 2018-07-07 by RMerl

nullcake, 2018-07-07:
The 'webmon' module for iptables is broken due to a missing lib. The kernel module 'ipt_webmon' is present and loads, but iptables needs the library to support this.

<pre>
admin@router:/# iptables -A web_monitor -p tcp -m webmon --max_domains 2000 --max_searches 2000
iptables v1.4.15: Couldn't load match `webmon':No such file or directory
</pre>

The issues I am having with http://merlin-ip/AdaptiveQoS_WebHistory.asp being blank is possibly due to something else. Looking in the forums again for related issue...