RMerl/asuswrt-merlin.ng #171: Let’s Encrypt for DDNS cannot work

hxsf, 2018-07-13:
some info in syslog.log

```
Jul 13 15:10:08 kernel: /usr/sbin/acme-client: https://acme-v01.api.letsencrypt.org/acme/new-authz: bad HTTP: 429
Jul 13 15:10:08 kernel: /usr/sbin/acme-client: transfer buffer: [{ "type": "urn:acme:error:rateLimited", "detail": "Error creating new authz :: too many failed authorizations recently: see https://letsencrypt.org/docs/rate-limits/", "status": 429 }] (189 bytes)
Jul 13 15:15:18 kernel: /usr/sbin/acme-client: https://acme-v01.api.letsencrypt.org/acme/new-authz: bad HTTP: 429
Jul 13 15:15:19 kernel: /usr/sbin/acme-client: transfer buffer: [{ "type": "urn:acme:error:rateLimited", "detail": "Error creating new authz :: too many failed authorizations recently: see https://letsencrypt.org/docs/rate-limits/", "status": 429 }] (189 bytes)
Jul 13 15:20:07 kernel: /usr/sbin/acme-client: https://acme-v01.api.letsencrypt.org/acme/new-authz: bad HTTP: 429
Jul 13 15:20:07 kernel: /usr/sbin/acme-client: transfer buffer: [{ "type": "urn:acme:error:rateLimited", "detail": "Error creating new authz :: too many failed authorizations recently: see https://letsencrypt.org/docs/rate-limits/", "status": 429 }] (189 bytes)
Jul 13 15:25:09 kernel: /usr/sbin/acme-client: https://acme-v01.api.letsencrypt.org/acme/new-authz: bad HTTP: 429
Jul 13 15:25:09 kernel: /usr/sbin/acme-client: transfer buffer: [{ "type": "urn:acme:error:rateLimited", "detail": "Error creating new authz :: too many failed authorizations recently: see https://letsencrypt.org/docs/rate-limits/", "status": 429 }] (189 bytes)
Jul 13 15:25:38 kernel: /usr/sbin/acme-client: /tmp/.le/www/.well-known/acme-challenge: -C directory must exist
Jul 13 15:25:42 kernel: /usr/sbin/acme-client: /tmp/.le/www/.well-known/acme-challenge: -C directory must exist
Jul 13 15:30:24 kernel: /usr/sbin/acme-client: https://acme-v01.api.letsencrypt.org/acme/challenge/QFD_QUkW3Z5O8JbQV8BkmoIf4cPBtydGHoOzYR17n2U/5631406776: bad response
Jul 13 15:30:24 kernel: /usr/sbin/acme-client: transfer buffer: [{ "type": "http-01", "status": "invalid", "error": { "type": "urn:acme:error:connection", "detail": "Fetching http://hxsf.asuscomm.com/.well-known/acme-challenge/F87LiSgrg03rcPR-CWMIj2FEyOZgs-KU6E5S4daaEZ0: Timeout during connect (likely firewall problem)", "status": 400 }, "uri": "https://acme-v01.api.letsencrypt.org/acme/challenge/QFD_QUkW3Z5O8JbQV8BkmoIf4cPBtydGHoOzYR17n2U/5631406776", "token": "F87LiSgrg03rcPR-CWMIj2FEyOZgs-KU6E5S4daaE
```

It is a message `Timeout during connect (likely firewall problem)`, and I try connect my router from www, it cannot be connect by port 80.

paldier, 2018-07-19:
default:8443

hxsf, 2018-07-20:
@paldier It is the error message when I used Let’s Encrypt for DDNS.
I know the web panel can only get by https(default is 8443).

jiqiren, 2018-08-08:
You can use port 443 by going to the “AiCloud 2.0” -> settings and changing the port to something like 2443. 

Save settings, then go to administration->system.  Change the https to 443 for WAN and LAN.

After saving - https without port works fine for the web panel

hxsf, 2018-08-08:
@jiqiren  I means that the feature "https with Let’s Encrypt" need use http request to ensure the domain(such as 'xxx.asuscomm.com')  is yours.
the log above  is show 'acme(a  Let’s Encrypt cli tool)' can not ensure it beause cannot reach the route with request.

so, web panel port and my problem are two different things.
But thank you for your comment. :)

nynhex, 2018-09-03:
Same problem here on latest build for AC6U

RMerl, 2018-09-10:
Let's Encrypt implemented is closed source and outside of my control.

closed 2018-09-10 by RMerl