RMerl/asuswrt-merlin.ng #172: Guest Network - Internet Access controls not working.

CHRONOMASTER, 2018-07-14:
**Issue:**
Guest Network internet access controls do not block internet access.

**Steps to repeat issue:**
In Guest Network page, create or modify a guest network and set Internet Access to Disabled and select save.

**Expected result:**
When a user is connected to a Guest Network and the Internet Access is disabled, the user should not be able to access anything outside the local network.

**Actual result:**
A user connected to the Guest Network can access the internet (Google, Facebook, etc.) when the internet access is set to disabled.

**Hardware:**
RT-AC68U

**Firmware:**
384.5 and 380.69_2
I have only tested this feature on these firmware versions and neither of them worked.

**Troubleshooting steps:**
I have done a factory reset of the setting, but this did not resolve the issue.

RMerl, 2018-07-14:
Please read the label closely.  It says Intranet, not Internet.

Working as intended.

closed 2018-07-14 by RMerl

CHRONOMASTER, 2018-07-14:
You are correct, sorry about that. Dyslexia strikes again.