RMerl/asuswrt-merlin.ng #174: Pick up kernel patch if it useful from linksys.

odkrys, 2018-07-23:
Linksys made some of patches for their ea9300 router which use bcm4908 chipset.
I expect if they are common and useful then you can pick them up.

https://drive.google.com/file/d/152f1woPac93EEa_Zc3BMbm9igaQZqUa4/view

RMerl, 2018-07-23:
ping @themiron

RMerl, 2018-07-23:
Will require a lot of work to analyze each of them.  Different kernel version, at least one of those I checked seemed to be already patched in a much more comprehensive way in Asus's own code, etc...

odkrys, 2018-07-23:
Yes, there is stem that seems to made by broadcom. It already applied on Asus's kernel.
But isn't the same 4.1.27 kernel on asus?

RMerl, 2018-07-23:
You're right, sorry (for some reason I had 4.1.21 in mind).

themiron, 2018-07-23:
I’ll take a look, thanks

closed 2019-01-12 by odkrys