RMerl/asuswrt-merlin.ng #177: ddns zoneedit needs longer timeout

brunnels, 2018-07-29:
It's working but I get system logs and warning in gui about it not working because firmware tries to update too often.  I don't see a place to configure this.

Here's what I get in the logs
```
Jul 29 05:02:58 watchdog: start ddns.
Jul 29 05:02:58 rc_service: watchdog 343:notify_rc start_ddns
Jul 29 05:02:58 start_ddns: update WWW.ZONEEDIT.COM zoneedit, wan_unit 0
Jul 29 05:02:58 ddns_update: ez-ipupdate: starting...
Jul 29 05:02:58 ddns_update: connected to dynamic.zoneedit.com (64.68.198.48) on port 80.
Jul 29 05:02:59 ddns_update: error processing request
Jul 29 05:02:59 ddns_update: server output: HTTP/1.1 200 OK^M Date: Sun, 29 Jul 2018 10:02:59 GMT^M Server: Apache/2.2.22 (Debian)^M X-Powered-By: PHP/5.4.45-0+deb7u5^M Vary: Accept-Encoding^M Content-Length: 93^M Connection: close^M Content-Type: text/html^M ^M <ERROR CODE="702" PARAM="600" TEXT="Minimum 600 seconds between requests" ZONE="mydomain.org">
Jul 29 05:02:59 ddns_update: asusddns_update: 1
Jul 29 05:02:59 ddns_update: exit_main
```

RMerl, 2018-08-17:
The DDNS client should only update on WAN IP changes, or after the configured forced refresh interval on the DDNS configuration page.  Make sure it wasn't your router repeatedly trying to authenticate with the wrong credentials, eventually causing that new error message.

brunnels, 2018-08-20:
I’m sure it’s not. This happens after a fresh reboot but will stop after
the request restriction has passed and a successful client update happens.
Maybe it’s a timing issue with reboot and the associated WAN status changes.

RMerl, 2018-09-10:
Should be taken care of with the recent re-implementation of DDNS through inadyn.

closed 2018-09-10 by RMerl